windowsair/wireless-esp8266-dap #1: Some error codes

Xiagb728, 2020-08-30:
main.c L:71:
esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCAL_11B | WIFI_PROTOCAL_11G | WIFI_PROTOCAL_11N);
no PROTOCAL. 'A' to 'O', PROTOCOL is true.

build error:
[100%] Building C object CMakeFiles/esp8266_dap.elf.dir/project_elf_src.c.obj
[100%] Linking CXX executable esp8266_dap.elf
esp-idf/lwip/liblwip.a(sys_arch.c.obj):(.literal.sys_init+0x14): undefined reference to `pthread_key_create'
esp-idf/lwip/liblwip.a(sys_arch.c.obj):(.literal.sys_thread_sem_init+0x4): undefined reference to `pthread_setspecific'
esp-idf/lwip/liblwip.a(sys_arch.c.obj):(.literal.sys_thread_sem_get+0x0): undefined reference to `pthread_getspecific'
esp-idf/lwip/liblwip.a(sys_arch.c.obj): In function `sys_init':
C:/user/user/data/softwareAndData/ordinary/esp8266/esp32_msys2_toolchain/msys32/home/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_ar
ch.c:446: undefined reference to `pthread_key_create'
esp-idf/lwip/liblwip.a(sys_arch.c.obj): In function `sys_thread_sem_init':
C:/user/user/data/softwareAndData/ordinary/esp8266/esp32_msys2_toolchain/msys32/home/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_ar
ch.c:533: undefined reference to `pthread_setspecific'
esp-idf/lwip/liblwip.a(sys_arch.c.obj): In function `sys_thread_sem_get':
C:/user/user/data/softwareAndData/ordinary/esp8266/esp32_msys2_toolchain/msys32/home/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_ar
ch.c:494: undefined reference to `pthread_getspecific'
collect2.exe: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/esp8266_dap.elf.dir/build.make:230：esp8266_dap.elf] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2163：CMakeFiles/esp8266_dap.elf.dir/all] 错误 2
make: *** [Makefile:150：all] 错误 2
make failed with exit code 2

What should I do?

windowsair, 2020-08-30:
您是否使用的是esp32的SDK?目前该项目只支持esp8266

Did you use esp32 SDK? The repo is currently only supports esp8266.

Xiagb728, 2020-08-30:
@windowsair 
国人吗？
I use the sdk of esp8266, I have compiled and downloaded "hello_world", it is normal.
It may be an error of the SDK. I have seen this kind of error elsewhere, and it will be fine after he rebuilds the

windowsair, 2020-08-30:
是的。可能是SDK的问题，或者检查下是不是menuconfig的问题。或者看看这个: 
[Linkage errors in SDK (GIT8266O-411)](https://github.com/espressif/ESP8266_RTOS_SDK/issues/840)

It may be a problem with the SDK, or check if it is a issue with menuconfig. See Also:
[Linkage errors in SDK (GIT8266O-411)](https://github.com/espressif/ESP8266_RTOS_SDK/issues/840)

Xiagb728, 2020-08-30:
@windowsair 
确实是SDK的问题，重新下了个完整包就好了
但是RAM不够了
c:/user/user/data/softwareanddata/ordinary/esp8266/esp32_msys2_toolchain/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../.
./../xtensa-lx106-elf/bin/ld.exe: esp8266_dap.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/user/user/data/softwareanddata/ordinary/esp8266/esp32_msys2_toolchain/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../.
./../xtensa-lx106-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/user/user/data/softwareanddata/ordinary/esp8266/esp32_msys2_toolchain/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../.
./../xtensa-lx106-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 84 bytes
collect2.exe: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/esp8266_dap.elf.dir/build.make:230：esp8266_dap.elf] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2092：CMakeFiles/esp8266_dap.elf.dir/all] 错误 2
make: *** [Makefile:150：all] 错误 2
make failed with exit code 2

windowsair, 2020-08-30:
这些问题和menuconfig的配置有很大关系.再次检查您的menuconfig是否配置了相关选项.

These problems have a lot to do with the configuration of menuconfig. Check again whether your menuconfig is configured with relevant options.

Xiagb728, 2020-08-30:
@windowsair 
我的SDK版本是3.3，你的呢？

windowsair, 2020-08-30:
已经找到问题了。我会尝试在下一个版本修复该问题。现在请使用旧版本的SDK:
https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.3-rc1

I got it. I will try to fix this issue in the next version. Now please use the old version of the SDK:
https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.3-rc1

Xiagb728, 2020-08-30:
@windowsair 
谢谢^_^

windowsair, 2021-04-06:
Hi,我决定跳过这个版本。目前主线的v3.4-rc版本性能有一定的提升，等到主线更新后我会切换到v3.4版本的SDK。

要关闭这个问题了，欢迎随时重新打开。

----

Hi, I decided to skip this version. The current v3.4-rc version of the mainline has some performance improvements, and I will switch to the v3.4 version of the SDK when the mainline is updated.

Closed. Please feel free to reopen it.

closed 2021-04-06 by windowsair