windowsair/wireless-esp8266-dap #3: 使用最新的SDK 编译成功后链接出错

Kevincoooool, 2020-09-22:
97354@Kevincoooool MINGW32 ~/ESP8266_RTOS_SDK/wireless-esp8266-dap
# make -j20
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-92-g8facf4c0
Compiler version: 5.2.0
Python requirements from D:/AiThinkerIDE_V1.0/msys32/home/97354/ESP8266_RTOS_SDK/requirements.txt are satisfied.
LD build/esp8266_dap.elf
D:/AiThinkerIDE_V1.0/msys32/home/97354/ESP8266_RTOS_SDK/wireless-esp8266-dap/build/esp8266\libesp8266.a(startup.o):(.literal.user_init_entry+0x10): undefined reference to `app_main'
D:/AiThinkerIDE_V1.0/msys32/home/97354/ESP8266_RTOS_SDK/wireless-esp8266-dap/build/esp8266\libesp8266.a(startup.o): In function `user_init_entry':
D:/AiThinkerIDE_V1.0/msys32/home/97354/ESP8266_RTOS_SDK/components/esp8266/source/startup.c:85: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [D:\AiThinkerIDE_V1.0\msys32\home\97354\ESP8266_RTOS_SDK/make/project.mk:510：/home/97354/ESP8266_RTOS_SDK/wireless-esp8266-dap/build/esp8266_dap.elf] 错误 1

windowsair, 2020-09-22:
Duplicate of #1  有时间再解决

Kevincoooool, 2020-09-22:
你用的是哪个版本的sdk编译的

windowsair, 2021-04-06:
Hi,我决定跳过这个版本。目前主线的v3.4-rc版本性能有一定的提升，等到主线更新后我会切换到v3.4版本的SDK。

要关闭这个问题了，欢迎随时重新打开。

----

Hi, I decided to skip this version. The current v3.4-rc version of the mainline has some performance improvements, and I will switch to the v3.4 version of the SDK when the mainline is updated.

Closed. Please feel free to reopen it.

closed 2021-04-06 by windowsair