windowsair/wireless-esp8266-dap #4: Is it stable?

caiguang1997, 2020-09-30:
Is it **stable** when I use **download** and **debug** function? I heard that usbip has been too long without maintenance.

这个东西**下载**和**调试**时候**稳定**么？我听说usbip很久没人维护了。

windowsair, 2020-09-30:
是否稳定**不敢保证**，可以自己试一下
usbip windows旧版本用着问题不大，现在有人在积极维护，新版项目在这里：[usbip-win ](github.com/cezanne/usbip-win )
本项目在这个版本的usbip下运行存在一些问题，我会在下个大版本一起修复

I can **not guarantee** stability. You can try it yourself.
There should be no problems running under the old version. And now someone is actively maintaining new verison, it can be  found here: [usbip-win ](github.com/cezanne/usbip-win )
There are some issues in our repo running with this version of usbip, I will fix them in the next major release.

caiguang1997, 2020-09-30:
我打算测测看。你还记得它能达到一个什么样的速度，比如可以达到20kB/s

I plan to test it soon. Do you remember how fast can it reach? For example, it can reach a speed of 20kB/s.

windowsair, 2020-09-30:
好吧，速度测试这方面我属实搞不大清楚。似乎有多种衡量标准。。。您知道有啥测试方法么:disappointed_relieved:

Well, I'm not really sure about this aspect of speed testing. There seem to be multiple measures. Do you know of any testing methods?:disappointed_relieved:

caiguang1997, 2020-10-02:
😀想跟您讨论几个细节，可以加下联系方式么，我QQ（或者微信都）是2240726829

😀Can I communicate with you to discuss some details? Both my QQ and Wechat number are 2240726829

windowsair, 2020-10-04:
下面是我使用Keil进行测试的结果（232 kb bin 文件，进行擦除、下载、验证的用时）:
1. st-link 4MHz时钟 平均用时18s
2. 本项目 5MHz时钟 平均用时42s

:flushed: 确实是慢了点，还有改进的空间

Here are the results of my tests with Keil (232 kb bin file, time to erase, download, verify):
1. st-link 4Mhz clock : average 18s 
2. The project's 5MHz clock : average 42s

:flushed: it is indeed a bit slow, there is still room for improvement!

windowsair, 2021-03-08:
通过usbip对st link v2进行了测试，本项目要优于st link的表现。(之前我一直以为性能瓶颈是esp8266的tcp传输速度)

现在，需要等待上游usbip项目的更新。😀

The ST Link v2 was tested via usbip and result shows that this project outperformed the st link performance. (I always thought that the performance bottleneck was the tcp transmission speed of esp8266.)

Now, it is necessary to wait for the update of the upstream usbip project.😀

windowsair, 2021-03-08:
我可能需要进一步对jlink进行测试😀

I may need to do further testing on jlink...😀