windowsair/wireless-esp8266-dap #11: swo串口

Zy19930907, 2021-02-14:
大佬可以把串口加上吗？

windowsair, 2021-02-14:
@Zy19930907 :
感谢您提起这个。
目前IO资源比较紧张，短时间内没有加入SWO的计划。

Thanks for raising this.
 Currently IO resources are limited and there are no plans to join SWO anytime soon.

windowsair, 2021-02-14:
不过我们仍然可以通过IO复用的方式实现这一点。目前正在准备发布一个新版本，在这个版本完成之后，应该可以考虑加入SWO. 😂

But we can still achieve this by IO reuse. A new release is being prepared, and after this version is completed, we should be able to consider adding SWO.  😂

Zy19930907, 2021-02-14:
谢谢支持，我这边按照您的工程移植了一个Arduino版的工程到esp32上，现在esp32 IO比较充足，您这边考虑支持esp32吗？

windowsair, 2021-02-14:
有的，我们是有这个打算。您已经移植完毕了吗？进展如何呢😊

Yes, we do intend to do that. Have you finished porting yet? How is it going?😊

Zy19930907, 2021-02-14:
目前移植到esp32，Arduino工程，swd调试正常，我本来是想发给您的，有用的话可以参考一下，但是好像您的github邮箱拒收了，工程17M，您可以提供一个其他的邮箱我发给您吗？

windowsair, 2021-02-14:
辛苦了，这个移植的工作量不小。🥰

您可以直接发到msdn_01#sina.com的邮箱（QQ同号）

Yecgaa1, 2021-03-07:
> 目前移植到esp32，Arduino工程，swd调试正常，我本来是想发给您的，有用的话可以参考一下，但是好像您的github邮箱拒收了，工程17M，您可以提供一个其他的邮箱我发给您吗？
> […](#)

我看到您对这个仓库的分支https://github.com/Zy19930907/wireless-esp8266-dap，请问您这个仓库是您已经成功移植到Esp32了吗

windowsair, 2021-03-07:
Hi，@Yecgaa1

感谢您的关注，我已经和这位作者联系上了，他的工程暂时还未发布。

本项目的移植工作也正在进行中，可能不久之后发布。

Yecgaa1, 2021-03-07:
好的，我非常期待这个移植可以突破由于tcp传输速度瓶颈所导致的性能瓶颈

windowsair, 2021-03-08:
> 好的，我非常期待这个移植可以突破由于tcp传输速度瓶颈所导致的性能瓶颈

@Yecgaa1 关于稳定性和性能请查看这个线程： #4 

esp32带来的性能提升可能感知不强。。。

Yecgaa1, 2021-03-08:
> > 好的，我非常期待这个移植可以突破由于tcp传输速度瓶颈所导致的性能瓶颈
> 
> @Yecgaa1 关于稳定性和性能请查看这个线程： #4
> 
> esp32带来的性能提升可能感知不强。。。

我看到了你最新的测试成果
或许你可以尝试做一下发送器，或许可以绕过usbip的限制。虽然我可以猜到这其中的难度
当然，我仍然期待esp32可以带来一定提升，当然这需要看测试结果

Yecgaa1, 2021-03-09:
同时对esp32的适配也有利于swo的加入

kaide6, 2021-08-02:
> 大佬可以把串口加上吗？

大佬，烧录成功了吗，我现在连原文件还没编译成功，可以发一份，编译好的文件给我吗

windowsair, 2021-08-02:
> 现在还没有串口，编译好的版本可以看看release页面里的https://github.com/windowsair/wireless-esp8266-dap/releases
> 
> > 大佬可以把串口加上吗？
> 
> 大佬，烧录成功了吗，我现在连原文件还没编译成功，可以发一份，编译好的文件给我吗

现在还没有串口，编译好的版本可以看看release页面里的https://github.com/windowsair/wireless-esp8266-dap/releases

kaide6, 2021-08-02:
> > 现在还没有串口，编译好的版本可以看看release页面里的https://github.com/windowsair/wireless-esp8266-dap/releases
> > > 大佬可以把串口加上吗？
> > 
> > 
> > 大佬，烧录成功了吗，我现在连原文件还没编译成功，可以发一份，编译好的文件给我吗
> 
> 现在还没有串口，编译好的版本可以看看release页面里的https://github.com/windowsair/wireless-esp8266-dap/releases

好的感谢，刚开始接触，是在download中选择bootloadder.bin和
esp8266dap_app.bin就可以了吗
[image]

windowsair, 2021-08-02:
@kaide6 用full的，从0x00开始烧录

windowsair, 2021-08-02:
@kaide6 可以参考下这个
[image]

kaide6, 2021-08-02:
> @kaide6 可以参考下这个
> [image]
是这样吗？
[image]

windowsair, 2021-08-02:
@kaide6  上面的没问题，下面的flash size和spi选项需要根据板子来选

kaide6, 2021-08-02:
有群或者B站号吗？

windowsair, 2021-08-02:
> 有群或者B站号吗？

没有，可以在github上问，建议新开个issue，或者在discussion里讨论

windowsair, 2022-01-18:
朋友们好，暂时不考虑加入SWO功能了。不过有加入其他形式串口的打算， 我们在 #21 中进行详细讨论。

Hi all, SWO feature will not be added in the future. But a lightweight uart bridge is possible. See:  #21

closed 2022-01-18 by windowsair