windowsair/wireless-esp8266-dap #13: 我在复杂的网络条件下使用这个无线调试器，我希望自定义它的ip

Yecgaa1, 2021-03-07:
我在短暂的分析代码后没有发现这个功能
希望您可以帮我一下

windowsair, 2021-03-07:
您需要的是静态ip功能。我们会尽快加入这个功能。

What you need is the static ip feature. We will add this feature as soon as possible.

windowsair, 2021-03-07:
我想确认下您所说的自定义ip功能。是我们常说的静态ip功能吗（可能只适用于某个网关下的网段），还是具备什么其他的特性？

I would like to confirm the custom ip feature you are talking about. Is it the static ip feature we often refer to (which may only apply to segments under a certain gateway), or does it have any other features?

Yecgaa1, 2021-03-07:
是的，静态ip功能就可以了
因为当我有多个您方案中的无线调试器在同一网络中时，我希望我可以很好的区分它们，并且降低在寻找ip所需要的时间

windowsair, 2021-03-09:
https://github.com/windowsair/wireless-esp8266-dap/commit/8429361cf8fe5c958d095ca152431f8425a77be1

更新了。现在可以在`wifi_configuration.h`修改IP配置。

不过我更建议直接使用路由器提供的静态地址绑定功能（应该是个路由器就有的= = ）

------

Updated. Now you can change the IP configuration in `wifi_configuration.h`

However, I think you should use the static address binding feature that comes with most routers.

Yecgaa1, 2021-03-09:
确实调整路由器就可以达到这个效果
但是当我没有路由器管理权，或者低端路由器没有此功能时，固定ip是个更有效率的方案
非常感谢你

closed 2021-03-09 by Yecgaa1