windowsair/wireless-esp8266-dap #18: using a wemos d1 mini to flash this firmware but wifi didn't show up

xiezhuojin, 2021-05-14:
using command ```esptool write_flash 0 /home/kevin/storinator_beta/@documents/hardware/esp8266/esp8266_dap.bin``` to flash the firmware to my wemos d1 mini, but the WIFI didn't show up. A help will be graceful.

windowsair, 2021-05-14:
Hi, @xiezhuojin
This is not a full firmware. You need Espressif’s bootloader.

windowsair, 2021-05-14:
Sorry, I forgot about that. Currently you can follow the readme instructions for download  via idf.py, or you can find the appropriate bootloader to download along with the app firmware.

I will support an easier burning method ASAP.😢

windowsair, 2021-05-14:
Hi， @xiezhuojin 

You can try using the pre-release version of the following page:

https://github.com/windowsair/wireless-esp8266-dap/releases/tag/v0.1.3-alpha

Using `esp8266_dap_full.bin`, download it with the same command

xiezhuojin, 2021-05-14:
Tried to use the esp8266_dap_full.bin firmware with the same command, but it still didn't work. But thank you anyway, this is the first time I tried esp8266 chip.

windowsair, 2021-05-14:
It's weird. Are you able to see any output through the serial port with  baudrate of 74880?

xiezhuojin, 2021-05-14:
Here is the command and its output to flash to firmware
```
esptool  -p /dev/ttyUSB0 write_flash -fm dio 0 /home/kevin/storinator_beta/@documents/hardware/esp8266/esp8266_dap_full.bin
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:82:3c:cd
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Flash params set to 0x024f
Took 1.72s to erase flash block
Wrote 380928 bytes at 0x00000000 in 37.2 seconds (82.0 kbit/s)...

Leaving...
Hard resetting via RTS pin...
```

And there is the output of picocom
```
picocom --baud 74800 /dev/ttyUSB0 
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 74800
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
```

Please point out the place I did wrong, thanks.

windowsair, 2021-05-14:
The baud rate is 74880. We only concerned with the output results during power-on. Maybe you should  manually press the reset button on your board.

xiezhuojin, 2021-05-15:
Output of serial port using 74880 baud
``` 
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 6012, room 16 
tail 12
chksum 0x06
ho 0 tail 12 room 4
load 0x3ffe8008, len 24, room 12 
tail 12
chksum 0xae
ho 0 tail 12 room 4
load 0x3ffe8020, len 3320, room 12 
tail 12
chksum 0xe5
csum 0xe5
user code done
```

windowsair, 2021-05-15:
Can you please try to execute erase_flash before downloading?

`user code done` may mean that the flash mode or flash size needs to be changed.

windowsair, 2021-05-15:
ok, wemos d1 mini is an ESP-01/07 series chip. You may need to use qio flash mode:

```bash
esptool  -p /dev/ttyUSB0 write_flash -fm qio 0x00 path_to_bin
```

xiezhuojin, 2021-05-15:
esptool said that chip doesn't support erase_flash. After using qio mode, led blinks but the wifi still doesn't show up. Output of serial 
```
load 0x40100000, len 6012, room 16 
tail 12
chksum 0x06
ho 0 tail 12 room 4
load 0x3ffe8008, len 24, room 12 
tail 12
chksum 0xae
ho 0 tail 12 room 4
load 0x3ffe8020, len 3320, room 12 
tail 12
chksum 0xe5
csum 0xe5
```
Maybe I should buy a daplink...

windowsair, 2021-05-15:
Ok，it bring you one step closer to success.. I'll upload a firmware later that shows the full debug log.

windowsair, 2021-05-15:
[target.zip](https://github.com/windowsair/wireless-esp8266-dap/files/6486533/target.zip)

xiezhuojin, 2021-05-15:
After reading the log, it said ```Disconnect reason : 201```. So I try to look at code, and found the reason(maybe). I originally though the dap is trying to create an AP and my PC should connect to the AP to get it work. But instead, this dap is trying to act as a WIFI client to connect to AP call 'DAP', then I created a hotspot with my phone, the dap connected to it and seems everything work perfectly! Thanks a lot, such a great job you did for leading me(not even a hardware guy) to make this to the end!

closed 2021-05-15 by xiezhuojin

windowsair, 2021-05-15:
Yes. Most laptops have mobile hotspots which is the reason why the DAP acts as an STA to connect to the PC(AP).

I didn't think it would be ambiguous, thanks for pointing that out, I'll update the description accordingly in the next release.🤗