windowsair/wireless-esp8266-dap #19: 下载讨论

kaide6, 2021-08-02:


kaide6, 2021-08-02:
我太菜了，连现成的都不会用，现在下载成功了，但是没有看到WiFi，就是网上最普通的node mcu eps8266板子

[image]

windowsair, 2021-08-02:
可以参考这个，spi 一般选 dio 或者qio
[image]

windowsair, 2021-08-02:
没有看到wifi？可能搞错了，esp8266不是做AP的

kaide6, 2021-08-02:
还是没有看到，我以为烧录完成后打开电脑或者手机WiFi就能看到DAP的网络，github有点卡，甚至加载不了

windowsair, 2021-08-02:
您可能没认真看README, esp8266是需要连接入一个名为DAP的热点，而不是本身充当一个名为DAP的热点

#18

kaide6, 2021-08-02:
哦，这样啊，我再试试。我电脑设置了对应的热点，esp8266的灯瞬间亮了，应该是成功了

kaide6, 2021-08-02:
[image]

windowsair, 2021-08-02:
这样就没啥问题了，usbip连接上应该就能用了

kaide6, 2021-08-03:
> 这样就没啥问题了，usbip连接上应该就能用了

usbip看对应的安装步骤都是英文的，不懂，然后找了个中文教程，出现以下情况，但是不能在keil中识别到下载器

[image]
[image]
[image]

kaide6, 2021-08-03:
> > 这样就没啥问题了，usbip连接上应该就能用了
> 
> usbip看对应的安装步骤都是英文的，不懂，然后找了个中文教程，出现以下情况，但是不能在keil中识别到下载器
> 
> [image]
> [image]
> [image]

kaide6, 2021-08-03:
大佬，是不是最后的USB IP没有安装正确

closed 2021-08-03 by kaide6

windowsair, 2021-08-03:
用的usbip版本不对，参考readme里的

kaide6, 2021-08-08:
> 用的usbip版本不对，参考readme里的
大佬，我安装readme里的UDF电脑会蓝屏，但是安装其它版本的usbip就不会，然而会出现上面图片版本不对的问题

windowsair, 2021-08-08:
蓝屏就是上游软件的问题了，如果要用其他版本的，需要在`dap_configuration.h`中禁用`USE_WINUSB `选项，不过这个版本已经不维护了