windowsair/wireless-esp8266-dap #20: The question about "RDDI-DAP Error"

LanT33, 2021-08-28:
连上设备后，第一次打开keil的调试器设置窗口，看到调试器连接成功
after connect to the DAP,The first time I open the keil debugger settings window, the debugger connection is successful.

[image]

然后关闭窗口，第二次打开设置窗口，显示如下错误
I close the setting window,and open it again,it show the error as the following picture:

[image]

之后无论开关几次，都显示上述错误。
The above error is displayed no matter how many times I turn it on and off.

请问应该如何解决？
How can I solve it？

LanT33, 2021-08-28:
已经尝试过减小Max Clock和重新插线，仍未解决问题
I have been attempted to reduce Max Clock and rewire, but they are not available.

windowsair, 2021-08-28:
试试1MHz左右的速度看看？

LanT33, 2021-08-28:
> 试试1MHz左右的速度看看？

重新开启测试模式后，程序运作正常，非常感谢你的开源！
Well,After reopen test mod,everything become find.Thank you for your open-source!

windowsair, 2021-08-28:
Hi, @LanT33 
感谢您的反馈，这种情况一般是电路连接问题，建议3v3（最好接上），gnd，sclk，swdio均使用较短的线连接
还有一种情况网络干扰，esp8266受5G wifi的影响很大，如果一直显示这个可以试试看断开usbip重新连接看看

windowsair, 2021-08-30:
我要关闭这个问题了，如果您有任何疑问，欢迎重新打开它

closed 2021-08-30 by windowsair