windowsair/wireless-esp8266-dap #21: 非SWO串口

ZhuYanzhen1, 2021-11-14:
您好，我刚在Issue中发现了关于支持SWO串口的问题。请问是否有计划通过使用ESP8266的烧录串口（即UART0），并在PC端枚举一个虚拟串口设备来实现无线串口的调试呢，期待您的回复，谢谢。

Hello, I just found out about the SWO serial port support in Issue. Is it possible to implement a wireless debug serial port by using the burn-in serial port of ESP8266 (UART0) and enumerating a virtual serial device on the PC side?

windowsair, 2021-11-14:
Hi, @ZhuYanzhen1 ，感谢您的支持

1. swo的计划暂时搁置，有打算加入虚拟串口，但是时间不确定。
2. 掉线问题是指的什么？在旧版本中，出现异常情况一般可以通过重新连接usbip解决，有一个例外是usbip未处理好某些行为，这个已经在develop分支中的`c67afb0`版本中解决

ZhuYanzhen1, 2021-11-15:
非常感谢您的答复。另外，我最近在使用KiCAD和ESP8285芯片制作基于这个项目的极小体积调试器，想问下能否加入到开发中来，将我制作的硬件作为此项目的硬件参考例子呢？

Thank you very much for your reply. Also, I have recently been working on a very small debugger based on this project using KiCAD and the ESP8285 chip, and I was wondering if I could join the development and use the hardware I created as a hardware reference example for this project?

windowsair, 2021-11-15:
嗯嗯，欢迎对项目的任何贡献☺️

windowsair, 2021-11-16:
对了，最近打算先实现下tcp uart bridge的方案，这个相比虚拟串口和swo更加轻量

ZhuYanzhen1, 2021-11-16:
> 对了，最近打算先实现下tcp uart bridge的方案，这个相比虚拟串口和swo更加轻量

方便告知一下用到的引脚这么分配吗，我可以直接做到硬件上

ZhuYanzhen1, 2021-11-16:
好的，可以顺便看看ESP8285的引脚。不出意外的话我应该会用ESP8285来进行小型化设计。

windowsair, 2021-11-16:
8285应该是pin-to-pin兼容的吧？这个问题应该不大。比较理想的方案是用U0    RX和U1TX

ZhuYanzhen1, 2021-11-16:
> 8285应该是pin-to-pin兼容的吧？这个问题应该不大。比较理想的方案是用U0 &nbsp; &nbsp;RX和U1TX

8285是pin2pin兼容的，程序也是直接兼容的。就是8285除了内置1MB Flash外还多出了GPIO9和GPIO10供用户使用。

windowsair, 2021-11-16:
> 8285应该是pin-to-pin兼容的吧？这个问题应该不大。比较理想的方案是用U0    RX和U1TX

嗯嗯，其实用模组直接做也行，后续有打算加上ota功能，1m的flash可能不太够

windowsair, 2021-11-16:
都行，ota也可以按需开启

  

  
那我还是做8266+片外Flash吧，主要是模组太大了而且板上焊板我不是很喜欢（强迫症患者）

windowsair, 2022-01-18:
Hi, @ZhuYanzhen1 给您讲讲最近的一些改动

1.  加上了OTA，我测了esp8266的1MB flash没有问题，但是不清楚8285能不能用
2.  加上了串口透传，需要用到U0 Rx 和 U1 Tx
3.  我想砍掉LED功能，因为感知不强。 多出来的两个IO刚好可以给 nTRST  和 U1 Tx用

不知道您怎么看？期待您的回复😊

ZhuYanzhen1, 2022-01-18:
> Hi, @ZhuYanzhen1 给您讲讲最近的一些改动
>     1. 加上了OTA，我测了esp8266的1MB flash没有问题，但是不清楚8285能不能用
>     2. 加上了串口透传，需要用到U0 Rx 和 U1 Tx
>     3. 我想砍掉LED功能，因为感知不强。 多出来的两个IO刚好可以给 nTRST  和 U1 Tx用
> 不知道您怎么看？期待您的回复😊

&nbsp;&nbsp;&nbsp;&nbsp;我觉得挺好的，就是如果能保留一个LED用于指示工作状态就更好了，比如指示是否连接上WIFI。要不然在某些时候连不上设备时在查找故障方面会比较麻烦。另外就是想问一下OTA的文件是存在Github还是在哪里？是否需要提供文件服务器储存。

windowsair, 2022-01-18:
确实，原本dap的两个灯根本没啥用。那我还是把他改成显示wifi有没有连接吧，其他的状态好像也没啥必要？

关于ota，ota是针对项目本身的，文件就是编译好的固件，用户直接可以通过web进行更新
相关的更改情况在这里： https://github.com/windowsair/wireless-esp8266-dap/tree/3e3e3b6e686a36778213ecdba75ddbe7d4f98862#system-ota

ZhuYanzhen1, 2022-01-18:
> 确实，原本dap的两个灯根本没啥用。那我还是把他改成显示wifi有没有连接吧，其他的状态好像也没啥必要？
> 关于ota，ota是针对项目本身的，文件就是编译好的固件，用户直接可以通过web进行更新 相关的更改情况在这里： https://github.com/windowsair/wireless-esp8266-dap/tree/3e3e3b6e686a36778213ecdba75ddbe7d4f98862#system-ota

了解了，我还以为OTA是存放在服务器上，然后ESP8266连上WIFI之后自动检查更新&自动下载更新。其它LED状态确实没有什么必要，要有多的状态也可以用单个LED表示出来，比如单闪、双闪之类的。

windowsair, 2022-01-18:
噗，其实也是个自娱自乐的功能，OTA本身也没多少人用

那先这样，原先LED GPIO15的改为显示wifi连接状态； LED GPIO2 弃用，改为uart 1 Tx

ZhuYanzhen1, 2022-01-18:
> 噗，其实也是个自娱自乐的功能，OTA本身也没多少人用
> 那先这样，原先LED GPIO15的改为显示wifi连接状态； LED GPIO2 弃用，改为uart 1 Tx

好的，我过段时间修改一下电路。另外就是我在想要是能加上一个通过网页或者别的不需要重新编译固件的手段修改WIFI名和密码的方法就好了。每天在实验室和宿舍之间跑有点麻烦，现在只能用手机开热点，但也不是非常方便。

windowsair, 2022-01-18:
这个想法不错，加个wifi列表就好啦，我来整整看

ZhuYanzhen1, 2022-01-18:
> 这个想法不错，加个wifi列表就好啦，我来整整看

好，期待成果

windowsair, 2022-01-18:
🥰现在支持在wifi_configuration.h中添加多个wifi，会一个一个试直到找到合适的：

https://github.com/windowsair/wireless-esp8266-dap/commit/79e6de38281ed501546548f9acb4b83d898d23a6

ZhuYanzhen1, 2022-01-18:
> 🥰现在支持在wifi_configuration.h中添加多个wifi，会一个一个试直到找到合适的：
> [79e6de3](https://github.com/windowsair/wireless-esp8266-dap/commit/79e6de38281ed501546548f9acb4b83d898d23a6)

针不戳

closed 2024-06-09 by windowsair