windowsair/wireless-esp8266-dap #22: Which version of usbip are you using?

Iezuni, 2021-11-25:
Which version of usbip are you using?
我在使用usbip-win-0.3.5连接时,发现并没有usbip.exe attach_ude 这条命令并且报错

windowsair, 2021-11-25:
Hi, @lezuni

这是usbip-win的问题。在usbip-win 0.3.5中attach_ude 已经被废弃，直接用attch就行了

Iezuni, 2021-11-27:
使用命令 `.\usbip.exe attach -r 192.168.43.60 -b 1-1` 连接之后报错
`usbip: error: vhci driver is not loaded`

 esp8266 输出:
```
Socket accepted

Handling dev attach request...

Sending header...

Sending device info...

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending only first part of CONFIG

* GET 0x01 DEVICE DESCRIPTOR

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending only first part of CONFIG

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending ALL CONFIG

recv failed: errno 104
Shutting down socket and restarting...
```

windowsair, 2021-11-27:
> 使用命令 `.\usbip.exe attach -r 192.168.43.60 -b 1-1` 连接之后报错 `usbip: error: vhci driver is not loaded`
>
> esp8266 输出:
>
> ```
> Socket accepted
>
> Handling dev attach request...
>
> Sending header...
>
> Sending [...]

看样子是usbip驱动没有装，检查下设备管理器有没有这一项：
[image]

Iezuni, 2021-11-28:
没有这一项，，这个驱动是怎么安装呢？

windowsair, 2021-11-28:
具体还请参考usbip-win的文档，需要先安装证书，然后使用install选项安装驱动

Iezuni, 2021-11-28:
我刚刚已经安装了usbip的驱动，但是设备管理器的界面上显示了一个三角形 ，usbip install 显示已经成功安装 再次使用命令连接时 同样显示一样的错误 还有其他的办法吗

windowsair, 2021-11-28:
试下系统进测试模式再安装？

Iezuni, 2021-11-28:
虽然显示成功连接上了 但是不能在通用串行总线上找到esp8266😥 keil也没有cmsis dap设备

windowsair, 2021-11-28:
如果设备管理器看不到，就是没有连接上。
出于好奇，您用的是什么系统？

Iezuni, 2021-11-28:
刚刚没刷新 设备管理器上是有esp8266的 只是keil无法连接
系统是win10 20h2

windowsair, 2021-11-28:
能看看8266的串口输出吗？还有就是keil的版本是不是太旧了？

Iezuni, 2021-11-28:
这个是串口输出
```
Handling dev attach request...

Sending header...

Sending device info...

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending only first part of CONFIG

* GET 0x01 DEVICE DESCRIPTOR

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending only first part of CONFIG

* GET 0x02 CONFIGURATION DESCRIPTOR

Sending ALL CONFIG

* SET CONFIGURATION

* SET INTERFACE

* GET 0x0F BOS DESCRIPTOR

* GET MSOS 2.0 vendor-specific descriptor

** REQUESTED list of supported languages

* GET STATUS

** REQUESTED list of supported languages

```
keil 版本为5.25

windowsair, 2021-11-28:
我不太确定5.25是不是太旧了，网上好像有类似的问题，可以替换dll，或者更新keil来解决这个问题

windowsair, 2021-11-28:
我这边先试试看5.25能不能用吧

windowsair, 2021-11-28:
这是我测试的结果。值的注意的是，0.3.5版本的usbip如果重复使用attach命令进行连接，而没有显式使用detach断开连接的话，可能会出现一些奇怪的问题
[image]

Iezuni, 2021-11-30:
今天我再尝试的时候 在不关掉驱动程序强制签名时 usbip无法验证签名 在不验证签名时 我的电脑无法启动.....😰

windowsair, 2021-11-30:
可以试下进测试模式，这样不用每次重启都进那个禁用驱动强制要求签名的模式

Iezuni, 2021-11-30:
刚刚试了一下 我进入测试模式后会无限蓝屏

Iezuni, 2021-11-30:
我可能需要考虑使用其他的办法来实现那远程下载了

windowsair, 2021-11-30:
好吧= =这可能是系统问题，要不先把驱动卸载了，用旧的0.3.0版本的usbip？

windowsair, 2021-11-30:
0.3.0版本的安装是install_ude，连接是attch_ude，这个有输入对么😳

windowsair, 2021-11-30:
还要注意下安装前得手动卸载原有的旧驱动

Iezuni, 2021-11-30:
导致问题的应该是自签名的驱动 ,如果有办法解决驱动签名的问题估计才能解决问题....

windowsair, 2021-11-30:
如果真是签名问题，那可能暂时没办法了，上游的usbip项目暂时没有签名的计划😂

windowsair, 2022-06-23:
@Iezuni 

目前上游的usbip更新了0.3.6的开发版本，在这个版本中驱动已经经过微软的签名，无需安装证书。

https://github.com/cezanne/usbip-win/releases/tag/v0.3.6-dev

windowsair, 2022-12-14:
> @windowsair 我这边驱动也安装上了，设备管理器里也出现设备了，但是keil还是没认到 [image]
> 
> [image]

什么版本的usbip？ 串口有何输出？

JustQY, 2022-12-14:
最新的0.3.6

[image]

JustQY, 2022-12-14:
@windowsair 
更新后可以了。发现一个情况，必须设置频率为10Mhz才行

closed 2024-06-09 by windowsair