windowsair/wireless-esp8266-dap #23: build error

Huyil, 2021-12-05:
huyil233@Huyil:~/esp/ESP8266_RTOS_SDK/wireless-esp8266-dap-0.1.3.1/build2$ cmake ..
huyil233@Huyil:~/esp/ESP8266_RTOS_SDK/wireless-esp8266-dap-0.1.3.1/build2$ make all
[100%] Generating binary image from built executable
esptool.py v2.4.0
Generated /home/huyil233/esp/ESP8266_RTOS_SDK/wireless-esp8266-dap-0.1.3.1/build2/bootloader/bootloader.bin
[100%] Built target gen_project_binary
Scanning dependencies of target app
[100%] Built target app
[100%] No install step for 'bootloader'
[100%] Completed 'bootloader'
[100%] Built target bootloader
Scanning dependencies of target gen_project_binary
[100%] Generating binary image from built executable
esptool.py v2.4.0
.flash.rodata len 0x02ea4 load 0x4021b5ec

A fatal error occurred: Segment loaded at 0x4021b5ec lands in same 64KB flash mapping as segment loaded at 0x40210010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
make[2]: *** [CMakeFiles/gen_project_binary.dir/build.make:61: .bin_timestamp] Error 2
make[1]: *** [CMakeFiles/Makefile2:2847: CMakeFiles/gen_project_binary.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

windowsair, 2021-12-05:
Which version of sdk are you using？ Please try using the sdk that comes with the project.

Huyil, 2021-12-05:
ESP8266_RTOS_SDK_3.4

Huyil, 2021-12-05:
但是 ESP8266_RTOS_SDK_3.3rc 也会报错
but ESP8266_RTOS_SDK_3.3rc other error
huyil233@Huyil:~/esp/ESP8266_RTOS_SDK/wireless-esp8266-dap-0.1.3.1$ sudo python idf.py build
Setting IDF_PATH environment variable: /home/huyil233/esp/ESP8266_RTOS_SDK_3.3rc
Checking Python dependencies...
Python requirements from /home/huyil233/esp/ESP8266_RTOS_SDK_3.3rc/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running make in directory /home/huyil233/esp/ESP8266_RTOS_SDK_3.3rc/wireless-esp8266-dap-0.1.3.1/build
Executing "make -j 10 all"...
[  0%] Built target _project_elf_src
[  1%] Built target partition_table
[  1%] Built target esp8266_linker_script
[  2%] Performing build step for 'bootloader'
[  2%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c: In function 'load_partitions':
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c:208:66: error: argument to 'sizeof' in 'strncp
' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(item->info.label, (const char*) it->label, sizeof(it->label));
                                                                  ^
[  5%] Built target __idf_log
[  5%] Built target __idf_build_target
[  7%] Built target _project_elf_src
cc1: some warnings being treated as errors
[ 12%] Built target __idf_esp8266
make[2]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:63: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:3391: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 35%] Built target __idf_util
[ 72%] Built target __idf_bootloader_support
[ 75%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c: In function 'load_partitions':
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c:208:66: error: argument to 'sizeof' in 'strncp
' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(item->info.label, (const char*) it->label, sizeof(it->label));
                                                                  ^
cc1: some warnings being treated as errors
make[5]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:63: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:1937: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all] Error 2
make[3]: *** [Makefile:84: all] Error 2
make[2]: *** [CMakeFiles/bootloader.dir/build.make:112: bootloader-prefix/src/bootloader-stamp/bootloader-build] Error 2make[1]: *** [CMakeFiles/Makefile2:2720: CMakeFiles/bootloader.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2

windowsair, 2021-12-05:
There are some issues with the v3.4. Please try to compile with the sdk that comes with the project

windowsair, 2021-12-05:
项目根目录下包含了sdk，请用那个sdk进行编译

Huyil, 2021-12-05:
`idf.py  menuconfig` 成功了，但
`idf.py build` 失败
[image]

> ~/esp/ESP8266_RTOS_SDK_3.3/wireless-esp8266-dap-master$ idf.py build
Checking Python dependencies...
Python requirements from /home/huyil233/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running make in directory /home/huyil233/esp/ESP8266_RTOS_SDK_3.3/wireless-esp8266-dap-master/build
Executing "make -j 10 all"...
[  0%] Built target _project_elf_src
[  1%] Built target partition_table
[  1%] Built target esp8266_linker_script
[  1%] Performing build step for 'bootloader'
[  2%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c: In function 'load_partitions':
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c:208:66: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(item->info.label, (const char*) it->label, sizeof(it->label));
                                                                  ^
[  2%] Built target _project_elf_src
[  2%] Built target __idf_build_target
[  7%] Built target __idf_log
cc1: some warnings being treated as errors
make[2]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:63: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:3393: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 12%] Built target __idf_esp8266
[ 35%] Built target __idf_util
[ 72%] Built target __idf_bootloader_support
[ 75%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c: In function 'load_partitions':
/home/huyil233/esp/ESP8266_RTOS_SDK/components/spi_flash/src/partition.c:208:66: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
         strncpy(item->info.label, (const char*) it->label, sizeof(it->label));
                                                                  ^
cc1: some warnings being treated as errors
make[5]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:63: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/src/partition.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:1937: esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all] Error 2
make[3]: *** [Makefile:84: all] Error 2
make[2]: *** [CMakeFiles/bootloader.dir/build.make:112: bootloader-prefix/src/bootloader-stamp/bootloader-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:2721: CMakeFiles/bootloader.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2

windowsair, 2021-12-05:
请使用项目里的sdk进行编译，您现在用的不是项目里的sdk

Huyil, 2021-12-07:
谢谢，我成功了！！
SDK版本似乎还是v3.3rc？

> git clone https://hub.fastgit.org/espressif/ESP8266_RTOS_SDK.git
> cd ESP8266_RTOS_SDK
> git checkout 1be2289fcd68672f5d6bf6aafa4f4b57b20527a4
> git branch 1be2289fcd
> git checkout 1be2289fcd
> sudo pip install -r requirements.txt
>  . ./export.sh
> 
> git clone https://hub.fastgit.org/windowsair/wireless-esp8266-dap.git
>  cd wireless-esp8266-dap/
>  python3 idf.py build

windowsair, 2021-12-08:
是介于3.3和3.4之间的版本，这个版本的wifi性能表现稍微好点

closed 2021-12-21 by windowsair