windowsair/wireless-esp8266-dap #29: No bootable app partions in the partion table

ZhuYanzhen1, 2022-02-18:
&nbsp;&nbsp;&nbsp;&nbsp;运行`python idf.py build` & `python idf.py flash monitor`指令后报上述错误。

ZhuYanzhen1, 2022-02-18:
> 用的是哪个版本呢？

当前master分支的最新版本，散列值为：

windowsair, 2022-02-18:
你能不能试试ci自动编译的最新固件，我试了下是可以的：
https://github.com/windowsair/wireless-esp8266-dap/actions/runs/1818692237
https://github.com/windowsair/wireless-esp8266-dap/suites/5236140823/artifacts/160918901

ZhuYanzhen1, 2022-02-18:
> 你能不能试试ci自动编译的最新固件，我试了下是可以的： https://github.com/windowsair/wireless-esp8266-dap/actions/runs/1818692237 https://github.com/windowsair/wireless-esp8266-dap/suites/5236140823/artifacts/160918901

这里面有partion_table、bootloader、full和app共四个文件，是需要根据分区地址分别下载还是只需要把full下载至0x00地址处就行了

windowsair, 2022-02-18:
都可以。一般直接用full的，从0×00烧录就可以

ZhuYanzhen1, 2022-02-18:
> 都可以。一般直接用full的，从0×00烧录就可以

还是不行，一样的错误。我使用的是以下语句进行烧录：
`esptool.py`
你可以先erase flash然后再烧录看看能不能复现我这个错误

windowsair, 2022-02-18:
我是用GUI工具烧录的，先完全erase再烧录。你用的烧录命令是什么？

windowsair, 2022-02-18:
好的，我试试

windowsair, 2022-02-18:
我刚才试了下你的命令，没有问题= = 我再找找看是什么原因

ZhuYanzhen1, 2022-02-18:
> 我刚才试了下你的命令，没有问题= = 我再找找看是什么原因

我尝试了GUI工具，最终的Error是一样的，但是信息有一些不同：
```bash

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 5880, room 16
tail 8
chksum 0xeb
load 0x3ffe8008, len 24, room 0
tail 8
chksum 0x43
load 0x3ffe8020, len 3308, room 0
tail 12
chksum 0xdd
I (59) boot: ESP-IDF a7677d6 2nd stage bootloader
I (59) boot: compile time 14:51:08
I (68) qio_mode: Enabling default flash chip QIO
I (69) boot: SPI Speed      : 80MHz
I (72) boot: SPI Mode       : QIO
I (78) boot: SPI Flash Size : 16MB
I (85) boot: Partition Table:
I (90) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot: End of partition table
E (108) boot: No bootable app partitions in the partition table
user code done

```

windowsair, 2022-02-18:
能看看你的GUI烧录界面是怎么设置的吗？

windowsair, 2022-02-18:
额，有可能是硬件问题，你试试看esp8266_rtos_sdk里面的`examples\system\ota\native_ota\2+MB_flash\new_to_new_no_old` 例子
看看能不能用，排查一下是不是硬件的问题

windowsair, 2022-02-18:
只需要看看那个例子能不能正常引导就行

ZhuYanzhen1, 2022-02-18:
> 额，有可能是硬件问题，你试试看esp8266_rtos_sdk里面的`examples\system\ota\native_ota\2+MB_flash\new_to_new_no_old` 例子 看看能不能用，排查一下是不是硬件的问题

不能用，有可能确实是硬件的问题。不过很奇怪的是，使用GUI界面时能够读取出Flash的型号和大小，这让我不能确定是哪里的问题。

windowsair, 2022-02-18:
你用GUI烧录之前，有没有先点erase？

ZhuYanzhen1, 2022-02-18:
> 你用GUI烧录之前，有没有先点erase？

有，都是一样的报错。我再检查一下硬件，等它给的例程跑过再说。

windowsair, 2022-02-18:
还有，你可以考虑下更改flash spi mode 和 flash spi speed，在menuconfig里面改，有一些硬件可能上不到80MHz

ZhuYanzhen1, 2022-02-19:
我可能找到是什么原因了。我用的这版电路里面Flash QSPI引脚中有一对引脚的顺序是照着[数据手册](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_cn.pdf)画的，但它与[硬件设计文档](https://www.espressif.com/sites/default/files/documentation/esp8266_hardware_design_guidelines_cn.pdf)不同，有可能是这个原因。但是想要验证，需要等下一版板子到了焊接才知道。

windowsair, 2022-02-19:
嗯嗯，等新的硬件再看看

ZhuYanzhen1, 2022-02-25:
> 嗯嗯，等新的硬件再看看

解决了，就是上面的这个问题。我更新一下电路

closed 2022-02-25 by ZhuYanzhen1

windowsair, 2022-02-25:
好！能用就行