windowsair/wireless-esp8266-dap #31: OpenOCD连接失败

ZhuYanzhen1, 2022-02-25:
OpenOCD版本：Open On-Chip Debugger 0.11.0+dev-00571-g254883597-dirty (2022-02-15-15:05)，自己拉仓库下来编译的。ST Link等支持完美。
OpenOCD连接语句：`openocd.exe -f cmsis-dap.cfg`，cmsis-dap.cfg内容为`adapter driver cmsis-dap`。
连接的过程中就出错了，OpenOCD返回信息：
```bash
Open On-Chip Debugger 0.11.0+dev-00571-g254883597-dirty (2022-02-15-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf00a, serial=1234
Error: error writing data: Input/Output Error
Error: CMSIS-DAP command CMD_INFO failed.
```
串口返回的错误信息：
```bash
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (301) wifi:pm stop
Waiting for AP connection...
I (1905) wifi:state: 0 -> 2 (b0)
I (1958) wifi:state: 2 -> 3 (0)
I (1962) wifi:state: 3 -> 5 (10)
I (1984) wifi:connected with AcmeTech, aid = 1, channel 6, HT20, bssid = 2a:76:93:4d:4e:76
SYSTEM EVENT STA GOT IP : 192.168.1.171
Connected to AP
Socket created
Socket binded
Socket listening
Socket accepted
Handling dev attach request...
Sending header...
Sending device info...
* GET 0x02 CONFIGURATION DESCRIPTOR
Sending only first part of CONFIG
* GET 0x01 DEVICE DESCRIPTOR
* GET 0x02 CONFIGURATION DESCRIPTOR
Sending only first part of CONFIG
* GET 0x02 CONFIGURATION DESCRIPTOR
Sending ALL CONFIG
* SET CONFIGURATION
* SET INTERFACE
* GET 0x0F BOS DESCRIPTOR
* GET MSOS 2.0 vendor-specific descriptor
** REQUESTED list of supported languages
* GET STATUS
** REQUESTED list of supported languages
** REQUESTED list of supported languages
s2 handling cmd unlink...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4021f2db  PS      : 0x00000030  A0      : 0x40213d23  A1      : 0x3fffaeb0
A2      : 0x00000000  A3      : 0x3ffe906c  A4      : 0x00000204  A5      : 0xffffffff
A6      : 0x3fffaf38  A7      : 0x00000204  A8      : 0x3ffffff8  A9      : 0x3fffd824
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffe9060
A14     : 0x00000000  A15     : 0x3fffaf34  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
Backtrace: 0x4021f2db:0x3fffaeb0 0x40213d0f:0x3fffaee0
```
是不是不支持新版本的OpenOCD

windowsair, 2022-02-26:
core dump问题比较严重。用的是哪个版本的固件？如果方便的话，能不能上传编译之后的elf文件，路径应该是`build/esp8266_dap.elf`

windowsair, 2022-02-26:
相当严重的问题，我之前顾着搞其他功能去了，这里正经的dap全崩了。。。

windowsair, 2022-02-26:
找到原因了，栈太大炸了。。。

8266的内存实在有限，现在加了一堆功能，调小可能还会有暗病= =

windowsair, 2022-02-26:
试试看最新的开发分支，大概是没问题的

https://github.com/windowsair/wireless-esp8266-dap/commit/c13366dcd29fc98ad49015422190751314ef7269

ZhuYanzhen1, 2022-02-26:
> 试试看最新的开发分支，大概是没问题的
> 
> [c13366d](https://github.com/windowsair/wireless-esp8266-dap/commit/c13366dcd29fc98ad49015422190751314ef7269)
> 
> https://github.com/windowsair/wireless-esp8266-dap/actions/runs/1902137412

新的没有出现core panic的问题了。但是在连接目标MCU的时候还是会报错：
```bash
Open On-Chip Debugger 0.11.0+dev-00571-g254883597-dirty (2022-02-15-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use '
transport[0m select <transport>'.
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf00a, serial=1234
Info : CMSIS-DAP: SWD  supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 1234
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 1 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : stm32f4x.cpu: Cortex-M4 r0p1 processor detected
Info : SWD DPIDR 0x2ba01477
Error: Failed to write memory and, additionally, failed to find out where
Warn : target stm32f4x.cpu examination failed
Info : gdb port disabled
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Could not find MEM-AP to control the core
Error: Target not examined, reset NOT asserted!
embedded:startup.tcl:1136: Error: ** Unable to reset target **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1173
at file "embedded:startup.tcl", line 1136
Error: CMSIS-DAP command CMD_DISCONNECT failed.
```

复现方法：
1. 连接单片机，运行指令：`.\openocd.exe -f cmsis-dap.cfg -f stm32f4x.cfg`
   这两个文件在`openocd/share/openocd/scripts/interface`和`openocd/share/openocd/scripts/target`中

windowsair, 2022-02-26:
我手头没有M4的设备，但是M3测试是可以的

ZhuYanzhen1, 2022-02-26:
指令的返回依次是：
```
> init

> reset init
SWD DPIDR 0x2ba01477
SWD DPIDR 0x2ba01477
Could not find MEM-AP to control the core
Target not examined, reset NOT asserted!

> halt
Target not examined yet
```

windowsair, 2022-02-26:
openocd的连接命令是啥样的？

windowsair, 2022-02-26:
我感觉是板子没有选择对，因为这个MEM-AP是要和芯片匹配的。

我再找找看是什么原因

windowsair, 2022-02-26:
你这个`0x2ba01477` 我看着好像不是F4？像是国产的兼容板子？
然后网上说是CKS32，那你可能需要用F1的试试

windowsair, 2022-02-26:
我后面有时间找块f4试试看

windowsair, 2022-03-01:
我刚才试了块F4，没啥问题

不过8266确实很容易受到干扰

windowsair, 2022-03-02:
>  我觉得很有可能是我这个芯片的问题。不过ESP8266本身的收发能力也决定了它的稳定性和速度没法做到很高。
>
>
>
>  ---原始邮件---

还有个esp32的分支，有意思的是esp32的速度稍微比8266慢些，暂时还没有搞清楚是啥原因😂

ZhuYanzhen1, 2022-03-15:
Hi, 我再次遇到了这个问题，这次我使用的是MM32F3277G9。它是Cortex-M3内核的，而我使用ST-Link可以正常连接它。
看起来这并不是一个偶然现象。报错信息：
```bash
Open On-Chip Debugger 0.11.0+dev-00571-g254883597-dirty (2022-02-15-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use '
transport select <transport>'.
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,
disable}
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf00a, serial=1234
[0mInfo : CMSIS-DAP: SWD  supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 1234
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 1 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 10000 kHz
Error: Error connecting DP: cannot read IDR
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
```
OpenOCD配置文件：
```bash
adapter driver cmsis-dap

source [find target/swj-dp.tcl]
source [find mem_helper.tcl]

adapter speed 10000

if { [info exists CHIPNAME] } {
   set _CHIPNAME $CHIPNAME
} else {
   set _CHIPNAME stm32f1x
}

set _ENDIAN little

if { [info exists WORKAREASIZE] } {
   set _WORKAREASIZE $WORKAREASIZE
} else {
   set _WORKAREASIZE 0x1000
}

set _FLASH_SIZE 0x80000

if { [using_jtag] } {
   # See STM Document RM0008 Section 26.6.3
   set _CPUTAPID 0x3ba00477
} {
   # this is the SW-DP tap id not the jtag tap id
   set _CPUTAPID 0x2ba01477
}

swj_newdap $_CHIPNAME cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_CPUTAPID
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu

if {[using_jtag]} {
   jtag newtap $_CHIPNAME bs -irlen 5
}

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_m -endian $_ENDIAN -dap $_CHIPNAME.dap

$_TARGETNAME configure -work-area-phys 0x20000000 -work-area-size $_WORKAREASIZE -work-area-backup 0

set _FLASHNAME $_CHIPNAME.flash
flash bank $_FLASHNAME mm32f3x 0x08000000 $_FLASH_SIZE 0 0 $_TARGETNAME

adapter srst delay 100
if {[using_jtag]} {
 jtag_ntrst_delay 100
}

reset_config none

if {![using_hla]} {
    cortex_m reset_config sysresetreq
}

$_TARGETNAME configure -event examine-end {
	mmw 0xE0042004 0x00000307 0
}

$_TARGETNAME configure -event trace-config {
	mmw 0xE0042004 0x00000020 0
}
```

ZhuYanzhen1, 2022-03-15:
我怀疑是不是某一些openocd的配置使得它无法连接芯片？

windowsair, 2022-03-15:
这个比较玄学了。。因为第一步就发生了`cannot read IDR`错误，这是不能接受的。

实际上，第一步做的是读芯片的ID，这个基本上和openocd配置没有关系，因为Cortex M0、M1、M3、M7都是这样读的，调试器软件都是按这个规范来做的。如果说是高速下的时序问题，那改用低速方式就是原始的IO翻转，这里也不太可能出问题。

我还是觉得和硬件连接关系比较大；或者是usbip的问题，这个玩意在丢包条件下很不稳定

ZhuYanzhen1, 2022-03-15:
> 这个比较玄学了。。因为第一步就发生了`cannot read IDR`错误，这是不能接受的。
> 
> 实际上，第一步做的是读芯片的ID，这个基本上和openocd配置没有关系，因为Cortex M0、M1、M3、M7都是这样读的，调试器软件都是按这个规范来做的。如果说是高速下的时序问题，那改用低速方式就是原始的IO翻转，这里也不太可能出问题。
> 
> 我还是觉得和硬件连接关系比较大；或者是usbip的问题，这个玩意在丢包条件下很不稳定

我实在无法在硬件中找到什么问题。请问有没有办法不使用SP而I改用IO进行操作？

windowsair, 2022-03-16:
速度设置成小于10Mhz就是纯io翻转。

后续我具体测试下是哪里出现问题了

zjusocks, 2023-08-07:
STM32F303RE 开发版报错。

我只连接了SWD的SWCLK SWDIO GND 以及 RESET，同样的接线ST-LINK 通过OpenOCD是可以的。

```

parallels@ubuntu-linux-22-04-desktop:~/Downloads/DAP_TEST$ ~/Desktop/openocd-0.12/bin/openocd -f dap_test.cfg 
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Warn : Transport "swd" was already selected
adapter speed: 1000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf00a, serial=1234
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 1234
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: Error connecting DP: cannot read IDR
```

liushuai05, 2024-04-25:
> STM32F303RE 开发版报错。
>
> 我只连接了SWD的SWCLK SWDIO GND 以及 RESET，同样的接线ST-LINK 通过OpenOCD是可以的。
>
> ```
>
> parallels@ubuntu-linux-22-04-desktop:~/Downloads/DAP_TEST$ ~/Desktop/openocd-0.12/bin/openocd -f dap_test.cfg [...]

请教一下大佬linux是怎么通过openocd连上wifi上的esp的,我这边使用usbip挂载过来,一直连不上,openocd显示是连上了,但是是个死进程,然后telnet 4444也不通,并且openocd的进程杀不掉

closed 2024-06-09 by windowsair