windowsair/wireless-esp8266-dap #33: release的固件无法正常运行

yzl893, 2022-03-28:
硬件是nodemcu，开机打印信息：

[image]
下载设置：
[image]

yzl893, 2022-03-28:
修改SPI MODE为DIO后打印load partition table error
[image]

windowsair, 2022-03-28:
Hi， @yzl893 
下载固件的时候勾选Donotchgbin试试

yzl893, 2022-03-28:
> Hi， @yzl893 下载固件的时候勾选Donotchgbin试试

还是不行，似乎是因为手里的8266不支持QIO的缘故，但是下载的时候设置成DOUT好像也没有用
[image]

windowsair, 2022-03-28:
看样子没有烧录成功，可以多改几个烧录选项试试

LDEDA4587, 2022-10-12:
我的nodemcu，32Mbit，使用如下烧录设置：
[image]
上电启动后打印信息如下：
[image]
应该可以理解为烧录成功，但是此时如果按RST复位，将不能正常启动，打印信息如下：
[image]
如果再次上电启动，nodemcu可以正常连接热点，上位机也可以正常运行。

windowsair, 2022-10-12:
按RST后改变了某些引脚的电平，bootloader没有正常启动，而是进入到烧录模式

> 我的nodemcu，32Mbit，使用如下烧录设置： [image] 上电启动后打印信息如下： [image] 应该可以理解为烧录成功，但是此时如果按RST复位，将不能正常启动，打印信息如下： [image] 如果再次上电启动，nodemcu可以正常连接热点，上位机也可以正常运行。

closed 2024-06-09 by windowsair