windowsair/wireless-esp8266-dap #34: python3 ./idf.py build出问题

Alohaq, 2022-04-15:
esp-idf/main/libmain.a(main.c.obj):(.literal.mdns_setup+0x8): undefined reference to `mdns_init'
esp-idf/main/libmain.a(main.c.obj):(.literal.mdns_setup+0xc): undefined reference to `mdns_hostname_set'
esp-idf/main/libmain.a(main.c.obj):(.literal.mdns_setup+0x10): undefined reference to `mdns_instance_name_set'
esp-idf/main/libmain.a(main.c.obj): In function `mdns_setup':
/home/aloha/wireless-esp8266-dap/build/../main/main.c:51: undefined reference to `mdns_init'
/home/aloha/wireless-esp8266-dap/build/../main/main.c:58: undefined reference to `mdns_hostname_set'
/home/aloha/wireless-esp8266-dap/build/../main/main.c:66: undefined reference to `mdns_instance_name_set'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

windowsair, 2022-04-15:
大概率是sdk的submodule没有弄好。
参考在线编译的操作，使用项目里自带的sdk再编译试试

Alohaq, 2022-04-15:
好的 我再来看看 不过在这之前 还有 Executing "ninja all"...[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable esp8266_dap.elf
FAILED: esp8266_dap.elf

这两端话 不知道有没有影响

windowsair, 2022-04-15:
当然有影响。这是编译失败了

------------------ 原始邮件 ------------------
发件人: Alohaq ***@***.***&gt;
发送时间: 2022年4月15日 20:39
收件人: windowsair/wireless-esp8266-dap ***@***.***&gt;
抄送: windowsair ***@***.***&gt;, Comment ***@***.***&gt;
主题: 回复：[windowsair/wireless-esp8266-dap] python3 ./idf.py build出问题 (Issue #34)

Alohaq, 2022-04-15:
可以看出 是哪里的问题吗

windowsair, 2022-04-15:
“大概率是sdk的submodule没有弄好。
参考在线编译的操作，使用项目里自带的sdk再编译试试”

Alohaq, 2022-04-15:
好 谢谢！不过对我还是有点难 不是太懂。。。 不过 可不可以让这个程序在VScode下跑通呢？

windowsair, 2022-04-15:
这个问题我不知道要怎么回答= =

只要配置好工具链用什么都行

Alohaq, 2022-04-16:
重新安装了Ubuntu 应该和之前的ESP32的环境冲突了 下载成功了 现在连接成功后 不显示端口。。。

Alohaq, 2022-04-16:
只显示new usb device attached to usbvbus port 1 未显示 receive sequence

windowsair, 2022-04-16:
暂时没有测试过linux的环境，我改天看看吧

Alohaq, 2022-04-16:
就是Windows下的 是USBIP的原因吗

windowsair, 2022-04-16:
完整的错误信息？设备管理器能看到设备吗

Alohaq, 2022-04-18:
[image]
无法识别 这种情况是8266的原因 还是我驱动的原因

windowsair, 2022-04-18:
用的usbip版本太老了

Alohaq, 2022-04-20:
换成链接里的了 但是一直说证书没安装，可已经安装了  测试模式也已开启
[image]

Alohaq, 2022-04-20:
> 换成链接里的了 但是一直说证书没安装，可已经安装了 测试模式也已开启 [image]

如果选择手动安装，会提示系统不匹配
[image]

windowsair, 2022-04-20:
没遇到过这种情况，或许可以看看usbip那边的issue有没有类似问题

Alohaq, 2022-04-20:
好的 请问博主是用的win3.0版本吗？ 或许可以发我一份3.0的试一下？

windowsair, 2022-04-20:
https://github.com/cezanne/usbip-win/releases/tag/v0.3.0

Alohaq, 2022-04-20:
非常感谢您的耐心解答，不过我试了试，还是有问题，这个USBIP实在难搞 好不容易搞到一个能用的 好像和你的还不适配 这个项目先搁置了 大佬后续会出ESP32版本的吗  或者 会出一期讲解吗？

windowsair, 2022-04-21:
1. 很多人都会遇到usbip的问题，usbip本身没有稳定版本，这个没有办法。
2. 有esp32的版本，但是比较早了，没有同步这个项目的更新

bihailantian655, 2022-05-26:
> > 换成链接里的了 但是一直说证书没安装，可已经安装了 测试模式也已开启 [image]
> 
> 如果选择手动安装，会提示系统不匹配 [image]

同样的情况你解决了没 ，usbip  提交 issue  没人回复

windowsair, 2022-06-23:
@bihailantian655 @Alohaq 

目前上游的usbip更新了0.3.6的开发版本，在这个版本中驱动已经经过微软的签名，无需安装证书。

https://github.com/cezanne/usbip-win/releases/tag/v0.3.6-dev

VLa1111, 2022-07-01:
python3 idf.py build 报错flash映射冲突：
A fatal error occurred: Segment loaded at 0x4021b5ec lands in same 64KB flash mapping as segment loaded at 0x40210010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.

windowsair, 2022-07-01:
@Vla1111

大概率是sdk的submodule没有弄好。
参考在线编译的操作，使用项目里自带的sdk再编译试试

closed 2024-06-09 by windowsair