windowsair/wireless-esp8266-dap #35: vscode pio 配置

bihailantian655, 2022-05-16:
keil 倒是可以 

这个可以调试 stm  arduino 吧
platformio  怎么配置啊 

debug_tool = cmsis-dap
upload_protocol = cmsis-dap
upload_speed=1000

windowsair, 2022-05-16:
没有试过，但是应该也是通过openocd来调试的

https://docs.platformio.org/en/latest/plus/debug-tools/cmsis-dap.html#configuration

bihailantian655, 2022-05-16:
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, dfu, jlink, serial, stlink
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\genericSTM32F103CB\firmware.elf
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-10-16-21:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

swd
Error: timed out while waiting for target halted
embedded:startup.tcl:1070: Error: ** Unable to reset target **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1107
at file "embedded:startup.tcl", line 1070
*** [upload] Error 1

windowsair, 2022-05-16:
可能有点问题，我抽空看看

havenxie, 2022-05-16:
能否搞个Keil或者IAR的工程进行编译？

Haven
***@***.***

windowsair, 2022-05-16:
> 能否搞个Keil或者IAR的工程进行编译？ Haven ***@***.***  
> […](#)

什么工程？

bihailantian655, 2022-05-16:
我只用了swd（vcc gnd  sclk swo  rst），没用jtag      
是这个定义吧 
D0     STM32RESET
D1     TDI
D2     TDO
D3     TCK
D4     TMS
GND  GND
3.3V  3.3V

bihailantian655, 2022-05-16:
搞错了 ，我刷的别人的固件 。

bihailantian655, 2022-05-16:
https://blog.csdn.net/while0l1/article/details/90766151?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522165267347116782248550865%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D&request_id=165267347116782248550865&biz_id=0&utm_medium=distribute.pc_search_result.none-task-blog-2~all~first_rank_ecpm_v1~rank_v31_ecpm-1-90766151-null-null.142^v9^pc_search_result_cache,157^v4^control&utm_term=%E5%9F%BA%E4%BA%8EESP8266%E7%9A%84CMSIS-DAP%E8%B0%83%E8%AF%95%E5%99%A8&spm=1018.2226.3001.4187

windowsair, 2022-05-16:
嗯，这个确实不是这个项目的

bihailantian655, 2022-05-24:
github action   什么都没改  加了几个空格    编译报错

windowsair, 2022-05-24:
请提供相关的github action构建链接

windowsair, 2022-05-24:
已在最新版本中修复

bihailantian655, 2022-05-24:
> 请提供相关的github action构建链接
> […](#)

不懂这些 ，全部都是fork 你的仓库的。
action  workflow   进行到 merge bin  的时候报错。

bihailantian655, 2022-05-24:
我不知道是不是我个人的问题   电路中gpio15 需要再并联一个电阻 5k左右  ，否者不能正常启动 

esp12f    只是再 0x0 下载  esp8266_dap_full.bin    以后运行到这里就没反应了 

load 0x40100000, len 5880, room 16 
tail 8
chksum 0xeb
load 0x3ffe8008, len 24, room 0 
tail 8
chksum 0x43
load 0x3ffe8020, len 3308, room 0 
tail 12
chksum 0xd3
csum 0xd3
[0;32mI (44) boot: ESP-IDF ed19d9c 2nd stage bootloader[0m
[0;32mI (44) boot: compile time 05:11:34[0m
[0;32mI (44) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (50) boot: SPI Speed      : 80MHz[0m
[0;32mI (57) boot: SPI Mode       : QIO[0m
[0;32mI (63) boot: SPI Flash Size : 16MB[0m
[0;32mI (69) boot: Partition Table:[0m
[0;32mI (75) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (86) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (97) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (109) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (121) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
[0;32mI (132) boot:  4 ota_1            OTA app          00 11 00110000 000f0000[0m
[0;32mI (144) boot: End of partition table[0m
[0;32mI (150) boot: No factory image, trying OTA 0[0m
[0;32mI (158) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x592cc (365260) map[0m
[0;32mI (172) esp_image: segment 1: paddr=0x000692e4 vaddr=0x402692dc size=0x0ab80 ( 43904) map[0m
[0;32mI (185) esp_image: segment 2: paddr=0x00073e6c vaddr=0x3ffe8000 size=0x005fc (  1532) load[0m
[0;32mI (199) esp_image: segment 3: paddr=0x00074470 vaddr=0x40100000 size=0x00bd0 (  3024) load[0m
[0;32mI (213) esp_image: segment 4: paddr=0x00075048 vaddr=0x40100bd0 size=0x06ad0 ( 27344) load[0m
[0;32mI (226) boot: Loaded app from partition at offset 0x10000[0m

windowsair, 2022-05-24:
有点奇怪。试试这个能不能用
https://github.com/windowsair/wireless-esp8266-dap/releases/tag/v0.1.4

windowsair, 2022-05-24:
在调试的时候用的波特率是多少？

bihailantian655, 2022-05-24:
刷固件后重启 就卡在那里了，没有调试连不上wifi  ，这个固件也没有smartconfig吧。和电脑通讯是76800    115200.

windowsair, 2022-05-24:
正常应该是74880，但是区别不大

有很多用户反馈这个OTA分区的问题，但是我一直没法复现

或者也可以试试旧版本
https://github.com/windowsair/wireless-esp8266-dap/releases/tag/v0.1.3.1

bihailantian655, 2022-05-24:
能冒昧问一下这3个分区文件如果分别刷入的话 ，分别的刷入地址是什么么？
bootloader 
partiion_table
esp8266_dap_app

windowsair, 2022-05-24:
上面的版本已经没有OTA分区了，不太可能出现这样的结果

是怎么刷入的，能看看吗？

windowsair, 2022-05-24:
试试这个

[image]

bihailantian655, 2022-05-24:
我就是按照你的建议刷的。发第二张图片是想知道分别刷入的话地址应该填写多少

windowsair, 2022-05-24:
分开烧录结果应该是一样的

0x0  bootloader.bin 
0x8000 partition-table.bin
0x10000 esp8266_dap.bin

windowsair, 2022-05-24:
感觉很奇怪，或许可以擦除整块片子然后重新烧录试试，因为现在的分区表看着很多，感觉像没烧录进去的样子

bihailantian655, 2022-05-24:
我换了一个板子就变成这样了 ，但是连不上wifi
[image]

windowsair, 2022-05-24:
连不上WIFI只有两种可以
1. wifi是5g的
2. wifi ssid或者密码不对

可以检查一下

bihailantian655, 2022-05-24:
// #define WIFI_SSID "DAP"
// #define WIFI_PASS "12345678"

#define WIFI_SSID "MERCURY_4548"
#define WIFI_PASS "a2572135"

这个双引号 是要去掉么？ 

下面这个结构体要改么？

static struct {
    const char *ssid;
    const char *password;
} wifi_list[] __attribute__((unused)) = {
    {.ssid = "OTA", .password = "12345678"},
    {.ssid = "DAP", .password = "12345678"},
    // Add your WAP like this:
    // {.ssid = "your ssid", .password = "your password"},
};

windowsair, 2022-05-24:
```cpp
static struct {
    const char *ssid;
    const char *password;
} wifi_list[] __attribute__((unused)) = {
    {.ssid = "OTA", .password = "12345678"},
    {.ssid = "DAP", .password = "12345678"},
    {.ssid = "MERCURY_4548", .password = "a2572135"},
    // Add your WAP like this:
    // {.ssid = "your ssid", .password = "your password"},
};
```

bihailantian655, 2022-05-24:
原来是这样啊 ，这是一个wifi     ssid     密码    列表 。

#define WIFI_SSID "MERCURY_4548"
#define WIFI_PASS "a2572135"

static struct {
    const char *ssid;
    const char *password;
} wifi_list[] __attribute__((unused)) = {
    {.ssid = "OTA", .password = "12345678"},
    {.ssid = "DAP", .password = "12345678"},
    {.ssid = "MERCURY_4548", .password = "a2572135"},
    // Add your WAP like this:
    // {.ssid = "your ssid", .password = "your password"},
};

感谢 你这么耐心帮我

windowsair, 2022-05-24:
前面的定义忘记删了，，那个现在是没用的

smartconfig也不错，有时间我加一下这个功能

bihailantian655, 2022-05-24:
应该和我的wifi 名称的 下划线  有关系吧， 创建的 dap  的热点 能连接的上

windowsair, 2022-05-24:
不太清楚，至少我这里用特殊符号是没问题的

bihailantian655, 2022-05-24:
我用手机创建了一个同名 和密码的热点 能连接上，其他手机也能默认连接上 ，是不是固件设置 wifi  信号功率的问题啊。

windowsair, 2022-05-24:
应该不是

bihailantian655, 2022-05-25:
我好挫啊 怎么遇到这么多的问题呢 ，现在能连上wfi 但是 keil 不识别 ，中端接收不到信息。
这是两个板子分别刷了不同的固件的信息。互相替换固件一样的结果，都是arduino的固件可以用。
[image]
[image]
[image]
 。  用 arduino 编译的keil倒是可以识别 。

windowsair, 2022-05-25:
用的usbip版本不对，主页里有写

bihailantian655, 2022-05-25:
主页的那个我一装驱动就蓝屏。试过好几次。

windowsair, 2022-05-25:
那没办法，这两个驱动是不一样的

closed 2024-06-09 by windowsair