windowsair/wireless-esp8266-dap #39: command for esptool for flashing release binaries (not documented)

gitthnx, 2022-09-13:
It would be very helpful having an example for flashing esp8266 devices with esptool.py from command line, that is including all four new release binaries and related starting sectors. Thanks

windowsair, 2022-09-13:
I am not sure what you are going to do.  But, if you just want to write a complete bin, just do like this:
```bash
esptool.py --port <your port> write_flash 0x0 <xxx_full.bin>
```

The firmware with "full" in the release page is the complete firmware after merging, you don't need to care about the address information, just write it from 0x0 onwards.

gitthnx, 2022-09-14:
Thanks for the pointers.  

At the moment (14Sept2022, with newest git versions for ESP8266_RTOS_SDK_3.4, esp-idf and `idf.py build` command, and git version wireless-espX-dap) difficulties are with esp8266(12f), with serial output being non-readable and no possibility connecting to "DAP" wireless station, with precompiled release files (and errors on manually compiled repository on Linux Ubuntu 20.04).  

For wireless-esp32-dap, compiles with esp-idf toolchain (`idf.py build` from command line, with  
`./install.sh` and `. ./export.sh` from within esp-idf repository downloaded and same terminal window, for x86_64 and aarch64 platforms)  
and seems being flashed correctly with command suggested from esp-idf toolchain after successfully compiling  
`/root/.espressif/python_env/idf4.4_py3.8_env/bin/python ../../../root/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/esp32_dap.bin` 
(?1? merging files with `cat bootloader.bin partition-table.bin esp32-dap.bin > esp32-dap-all.bin` and from 0x0 then possible? tested, but only shows debug messages on serial console, like 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57 )
  
  
So far only separate flashing with separate start addresses for sections (bootloader, partition-table, esp32-dap binary) gets into a xtensa Kernel, but with `usbip list -r xxx.xxx.xxx.xxx` triggering reboot, because of cpu0 error
  

> Socket created
> Socket binded
> Socket listening
> Socket accepted
> Handling dev list request...
> Sending header...
> Sending device list...
> Sending device list size...
> Sending device info...
> Sending interface info...
> Connection closed
> Shutting down socket and restarting...
> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
> 
> Core  0 register dump:
> PC      : 0x40140762  PS      : 0x00060934  A0      : 0x800d2b4e  A1      : 0x3ffbbb80  
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x80000001  
> A6      : 0x3ff000e0  A7      : 0x00000001  A8      : 0x800e07f2  A9      : 0x3ffbbb50  
> A10     : 0x00000000  A11     : 0x40001d48  A12     : 0x800856f2  A13     : 0x3ffb0b40  
> A14     : 0x3ffb3a98  A15     : 0x3ffb3a98  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> 
> Backtrace:0x4014075f:0x3ffbbb800x400d2b4b:0x3ffbbba0 0x4008a755:0x3ffbbbc0 0x4008c415:0x3ffbbbe0 
> 
> 
> Core  1 register dump:
> PC      : 0x4008c578  PS      : 0x00060634  A0      : 0x8008bc0e  A1      : 0x3ffc7bf0  
> A2      : 0x3ffb11b8  A3      : 0xffffffff  A4      : 0x00000000  A5      : 0x00000000  
> A6      : 0x00000000  A7      : 0x00000451  A8      : 0x80088b7e  A9      : 0x0000abab  
> A10     : 0x00060623  A11     : 0x0000cdcd  A12     : 0xb33fffff  A13     : 0x00060623  
> A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000008  EXCCAUSE: 0x00000005  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> 
> Backtrace:0x4008c575:0x3ffc7bf00x4008bc0b:0x3ffc7c10 0x4008307e:0x3ffc7c30 0x4008c415:0x3ffc7c60

> ELF file SHA256: 17bb8bcf8757be72
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:6616
> load:0x40078000,len:14780
> ho 0 tail 12 room 4
> load:0x40080400,len:3792
> entry 0x40080694
> 

?2? How debugging this? Thx

windowsair, 2022-09-14:
1.) For esp8266, please use the SDK included in the project root directory, which contains some project-specific modifications. There is no guarantee for the effect of using other versions.

2.) wireless-esp32-dap is an earlier branch of this repo. Now, all esp32 changes will be ported to this repo.
2.1) The port on this repo is currently in progress and has not yet been tested.
2.2) If you really want to use the esp32 related features as soon ASAP, please use the latest version of the develop branch in wireless-esp32-dap. This version fixes the panic issue mentioned above.

2.3) The method you are using to merge the firmware is wrong. cat tool does appear to merge three firmwares, but the three firmwares are not contiguous and you need to fill the gaps in them with zeros or other data depending on the firmware address. This can be done using esptool, and if you do want to see examples, you can check the ci of this project, where there are operations related to merging firmware. Considering that you are using the SDK, you can use the more convenient idf.py tool, where you can simply use the idf.py flash to do the write operation.

gitthnx, 2022-09-14:
1.) it's `git clone --recursive https://github.com/windowsair/wireless-esp8266-dap`?
```
cd ESP8266_RTOS_SDK/
./install.sh 
. ./export.sh
```
recommends `make` (not idf.py) command (tested ok with examples/get-started/hello_world) 

```
cd wireless-esp8266-dap
idf.py set-target 
```
>Usage: idf.py set-target [OPTIONS] {esp32|esp32s2|esp32c3|esp32s3|linux|esp32h2}  

(no esp8266 target?)
```
idf.py -p /dev/ttyUSB1 build
```

> Executing "ninja all"...
> [1/4] cd /dev/shm/wireless-esp8266-dap/build/esp-idf/esptool_py && /root/.espressif/python_env/idf4.4_py3.8_env/bin/python /root/esp...app /dev/shm/wireless-esp8266-dap/build/partition_table/partition-table.bin /dev/shm/wireless-esp8266-dap/build/wireless_esp_dap.bin
> wireless_esp_dap.bin binary size 0xb0c10 bytes. Smallest app partition is 0x100000 bytes. 0x4f3f0 bytes (31%) free.
> [2/4] Performing build step for 'bootloader'
> [1/1] cd /dev/shm/wireless-esp8266-dap/build/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf4.4_py3.8_env/bin/python /root/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /dev/shm/wireless-esp8266-dap/build/bootloader/bootloader.bin
> Bootloader binary size 0x69a0 bytes. 0x660 bytes (6%) free.
> 
> Project build complete. To flash, run this command:
> /root/.espressif/python_env/idf4.4_py3.8_env/bin/python ../../../root/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/wireless_esp_dap.bin
> or run 'idf.py -p /dev/ttyUSB1 flash'

flashed to esp8266 (ESP-01, 1MB flash size and 12f, 4MB flash, 26MHz osc, 40MHz flash clk)
```
esptool.py -p /dev/ttyUSB1 -b 230400 --before default_reset --after hard_reset --chip esp8266  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partition-table.bin 0x10000 wireless_esp_dap.bin
```
```
idf.py -p /dev/ttyUSB1 monitor
```
>--- idf_monitor on /dev/ttyUSB1 115200 ---
>--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> rll�r$�n
> 
>         �
>          l�
>            b|���rb�
>                    b�nnlnnb
>                            bp�$blrlp�n0
> 
>                                        �
>                                         l
> 
> 
> 
> 
> 
>                                          b
>                                           n�nl
> 
> 
>                                               b��nn'l�l`�nn
>                                                            l`nr���n
> 
>                                                                    �
>                                                                     l`p�n�
> 
>                                                                           �l
>                                                                             Č
>                                                                              �l�b
>                                                                                  n�nl�
>                                                                                       ��
>                                                                                         l
>                                                                                          b��nn'
> 
>                                                                                                l`
?3? Someone seen this (tested common baudrates beside 115200, but no clear log output)?

2.1) [14Sept2022](https://github.com/windowsair/wireless-esp32-dap/issues/5)
2.2) tbd
2.3) seems `esptool` not yet includes [merge_image](https://github.com/espressif/esptool/issues/254), but maybe with version 3.x  
       [alternative esp32_binary_merger](https://github.com/vtunr/esp32_binary_merger) (not yet tested)

Thx

windowsair, 2022-09-14:
baudrate is 74880

gitthnx, 2022-09-14:
Ok, configurations starting to fit for hardware here.

can flash 'esp8266_dap_full.bin' on 0x0 now, 
```
esptool.py -p /dev/ttyUSB1 -b 921600 --before default_reset --after hard_reset --chip esp8266  write_flash --flash_mode dio --flash_size detect --flash_freq 80m  0x10000 /dev/shm/wireless-esp8266/esp8266_dap_app.bin
```

> load 0x40100000, len 5880, room 16 
> tail 8
> chksum 0xeb
> load 0x3ffe8008, len 24, room 0 
> tail 8
> chksum 0x43
> load 0x3ffe8020, len 3308, room 0 
> tail 12
> chksum 0xdb
> I (42) boot: ESP-IDF ed19d9c 2nd stage bootloader
> I (42) boot: compile time 02:21:38
> I (43) qio_mode: Enabling default flash chip QIO
> I (49) boot: SPI Speed      : 80MHz
> I (55) boot: SPI Mode       : QIO
> I (62) boot: SPI Flash Size : 4MB
> I (68) boot: Partition Table:
> I (73) boot: ## Label            Usage          Type ST Offset   Length
> I (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (96) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> I (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> I (119) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
> I (131) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
> I (142) boot: End of partition table
> I (149) boot: No factory image, trying OTA 0
> I (157) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x59548 (365896) map
> I (170) esp_image: segment 1: paddr=0x00069560 vaddr=0x40269558 size=0x0ab84 ( 43908) map
> I (184) esp_image: segment 2: paddr=0x000740ec vaddr=0x3ffe8000 size=0x005fc (  1532) load
> I (197) esp_image: segment 3: paddr=0x000746f0 vaddr=0x40100000 size=0x00bd0 (  3024) load
> I (211) esp_image: segment 4: paddr=0x000752c8 vaddr=0x40100bd0 size=0x06ad0 ( 27344) load
> I (225) boot: Loaded app from partition at offset 0x10000
> phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
> I (295) wifi:pm stop
> Waiting for AP connection...
> Disconnect reason : 201

but cannot continue with manually compiled 'esp8266_dap_app.bin' on 0x10000, although binary size <983040 bytes, actually is ~730kB
> I (42) boot: ESP-IDF ed19d9c 2nd stage bootloader
> I (42) boot: compile time 02:21:38
> I (43) qio_mode: Enabling default flash chip QIO
> I (49) boot: SPI Speed      : 80MHz
> I (55) boot: SPI Mode       : QIO
> I (62) boot: SPI Flash Size : 4MB
> I (68) boot: Partition Table:
> I (73) boot: ## Label            Usage          Type ST Offset   Length
> I (85) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (96) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> I (108) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> I (119) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
> I (131) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
> I (142) boot: End of partition table
> I (149) boot: No factory image, trying OTA 0
> I (157) esp_image: segment 0: paddr=0x00010010 vaddr=0x000000ee size=0x00000 (     0) load
> I (170) esp_image: segment 1: paddr=0x00010018 vaddr=0x00000000 size=0x1000000 (16777216) load
> I (185) esp_image: segment 2: paddr=0x01010020 vaddr=0x00000000 size=0x00000 (     0) load
> I (198) esp_image: segment 3: paddr=0x01010028 vaddr=0x00000000 size=0x00000 (     0) load
> I (212) esp_image: segment 4: paddr=0x01010030 vaddr=0x00000000 size=0x00000 (     0) load
> I (226) esp_image: segment 5: paddr=0x01010038 vaddr=0x00000000 size=0x00000 (     0) load
> E (240) esp_image: Image length 16777272 doesn't fit in partition length 983040
> E (252) boot: OTA app partition slot 0 is not bootable
> I (261) esp_image: segment 0: paddr=0x00110010 vaddr=0x000000ee size=0x00000 (     0) load
> I (274) esp_image: segment 1: paddr=0x00110018 vaddr=0x00000000 size=0x1000000 (16777216) load
> I (289) esp_image: segment 2: paddr=0x01110020 vaddr=0x00000000 size=0x00000 (     0) load
> I (303) esp_image: segment 3: paddr=0x01110028 vaddr=0x00000000 size=0x00000 (     0) load
> I (316) esp_image: segment 4: paddr=0x01110030 vaddr=0x00000000 size=0x00000 (     0) load
> I (330) esp_image: segment 5: paddr=0x01110038 vaddr=0x00000000 size=0x00000 (     0) load
> E (344) esp_image: Image length 16777272 doesn't fit in partition length 983040
> E (356) boot: OTA app partition slot 1 is not bootable
> E (365) boot: No bootable app partitions in the partition table
> user code done

Thx

windowsair, 2022-09-14:
I am not sure why you would want to use the esptool.py tool. Currently the project only follows the partition addresses in esp-idf for merging, any customization of partition addresses in a non esp-idf way is out the scope of this project

gitthnx, 2022-09-14:
If there's a possibility for customizing wifi configuration's **essid** and **password** there would be no need for recompiling 'esp8266_dap_app.bin' and flashing to 0x10000 being replacement for release binary with station id 'DAP' and 'OTA'.  
(Should i open another issue for that question?).  

(Bootloader@0x1000 and partition-table@0x8000 are suitable, once installed no change necessary and for flashing customized 'esp8266_dap_app.bin' esptool works flawlessly with having some experience now and sufficient power to esp modules.)  
  
IIUC, compilation of 'esp8266_dap_app.bin' has to be adjusted within esp-idf (idf.py menuconfig?, config files?, header files?) and there's no esptool (parameters) related influence to locating the binaries in segments (map?, load?, paddr?, vaddr?) from that point in compiling progress?

release binary 'esp8266_dap_app.bin'
```
esptool.py --chip esp8266 image_info esp8266_dap_app.bin 
```
> esptool.py v3.2-dev
> WARNING: Suspicious segment 0x40210010, length 365896
> WARNING: Suspicious segment 0x40269558, length 43908
> Image version: 1
> Entry point: 40211524
> 5 segments
> 
> Segment 1: len 0x59548 load 0x40210010 file_offs 0x00000008 [IROM]
> Segment 2: len 0x0ab84 load 0x40269558 file_offs 0x00059558 [IROM]
> Segment 3: len 0x005fc load 0x3ffe8000 file_offs 0x000640e4 [DRAM]
> Segment 4: len 0x00bd0 load 0x40100000 file_offs 0x000646e8 [IRAM]
> Segment 5: len 0x06ad0 load 0x40100bd0 file_offs 0x000652c0 [IRAM]
> Checksum: 20 (valid)
  
custom binary 'wireless_esp_dap.bin'
```
esptool.py --chip esp8266 image_info build/wireless_esp_dap.bin
```
> esptool.py v3.2-dev
> WARNING: Suspicious segment 0xee, length 0
> WARNING: Suspicious segment 0x0, length 16777216
> 
> A fatal error occurred: End of file reading segment 0x0, length 16777216 (actual length 723992)

custom binary 'wireless_esp_dap.bin'
```
esptool.py --chip esp32 image_info build/wireless_esp_dap.bin
```
> esptool.py v3.2-dev
> Image version: 1
> Entry point: 400815bc
> 6 segments
> 
> Segment 1: len 0x15cb4 load 0x3f400020 file_offs 0x00000018 [DROM]
> Segment 2: len 0x035e4 load 0x3ffb0000 file_offs 0x00015cd4 [BYTE_ACCESSIBLE,DRAM]
> Segment 3: len 0x06d50 load 0x40080000 file_offs 0x000192c0 [IRAM]
> Segment 4: len 0x80268 load 0x400d0020 file_offs 0x00020018 [IROM]
> Segment 5: len 0x10958 load 0x40086d50 file_offs 0x000a0288 [IRAM]
> Segment 6: len 0x00010 load 0x50000000 file_offs 0x000b0be8 [RTC_DATA]
> Checksum: 03 (valid)
> Validation Hash: 98c0d6acbe4843a34da32f7c4b388489d5063041a5df617f58d151063a9284cc (valid)

Recognized with summarizing, there seems being a wrong configuration within esp-idf for esp32 instead of esp8266 binaries, but there's was/is no option for choosing esp8266 or esp32 device target type (only esp32 variants are offered with `idf.py set-target`)?

Thx

windowsair, 2022-09-14:
esp NVS may be what you are looking for

windowsair, 2022-09-14:
okay, the hack you want to do may be a bit complicated, at the very least you have to make sure the partition is not overwritten.

I don't have much experience with partition configuration, but if it were up to me to implement this feature, I'd be inclined to just leave it to the compiler to do the work, and the compiler would open up about 100byte of space, which I'd probably fill with special garbage data, so I could quickly find the starting location of the address in binary form.

gitthnx, 2022-09-14:
(For clarifying: Some of previous errors and confusion was from `echo $IDF_PATH` variable was configured for esp32 devices from earlier esp-idf Framework installation, what was not automatically adjusted with `. ./export.sh` from following ESP8266_RTOS_SDK installation process. So all binaries compiled with xtensa-esp32 toolchain instead of xtensa-lx106 and curiously finished without errors. Another task was downgrading python cryptography to <35.0 with `pip3 install cryptography==3.4.8`.
Converting .elf file to .bin is now only remaining difficulty.
> A fatal error occurred: Segment loaded at 0x4021afcc lands in same 64KB flash mapping as segment loaded at 0x40210010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
Thx )

closed 2022-09-17 by gitthnx