windowsair/wireless-esp8266-dap #41: error on generating binary on Ubuntu 20.04 distribution: .flash.rodata len 0x02ec4 load 0x4021b5ec

gitthnx, 2022-09-17:
sys: x86_64 
os: Linux, Ubuntu 20.04
sdk: Project version: v3.4-dev-347-g1be2289f (change from `make` to `idf.py` command since version [v0.1.3](https://github.com/windowsair/wireless-esp8266-dap/releases/tag/v0.1.3))
wireless-esp8266-dap tags: (comparable resulting compile error) v0.1.0-v0.2.0

```
idf.py -p /dev/ttyUSB1 app
```
e.g. from v0.1.3.1
> [578/595] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/tcp_server.c.obj
> ../main/tcp_server.c: In function 'tcp_server_task':
> ../main/tcp_server.c:148:32: warning: this statement may fall through [-Wimplicit-fallthrough=]
>                          kState = ATTACHING;
>                          ~~~~~~~^~~~~~~~~~~
> ../main/tcp_server.c:150:21: note: here
>                      case ATTACHING:
>                      ^~~~
> [595/595] Generating binary image from built executable
> FAILED: .bin_timestamp 
> cd /dev/shm/dap-0.3.1/build && /root/.espressif/python_env/rtos3.4_py3.8_env/bin/python /mnt/nvme0n1p2/Stm32/cmsisDap/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py --chip esp8266 elf2image --flash_mode dio --flash_freq 80m --flash_size 16MB --version=3 -o /dev/shm/dap-0.3.1/build/esp8266_dap.bin esp8266_dap.elf && /usr/bin/cmake -E echo "Generated /dev/shm/dap-0.3.1/build/esp8266_dap.bin" && /usr/bin/cmake -E md5sum /dev/shm/dap-0.3.1/build/esp8266_dap.bin > /dev/shm/dap-0.3.1/build/.bin_timestamp
> esptool.py v2.4.0
> .flash.rodata len 0x02ec4 load 0x4021b5ec
> 
> A fatal error occurred: Segment loaded at 0x4021b5ec lands in same 64KB flash mapping as segment loaded at 0x40210010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

(momentarily workaround for debugging: compiles with copying sdkconfig from compilable ESP8266_RTOS_SDK/examples/get-started/hello_world to 'wireless-esp8266-dap' root directory, but resulting binary not yet functional for Debug Access Port (DAP) task, stalls with bootloader and partition-table flashed on
>W (166) esp_image: image at 0x10000 has invalid SPI mode 67
>***
> E (203) boot: No bootable app partitions in the partition table
> user code done  

`diff -y sdkconfig_helloWorld sdkconfig`

```
CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG=y		     |	# CONFIG_COMPILER_OPTIMIZATION_LEVEL_DEBUG is not set
# CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE is not set	     |	CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE=y
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE=y	     |	# CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE is not set
# CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE is not se  |	CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE=y
# CONFIG_APP_UPDATE_CHECK_APP_HASH is not set		     |	CONFIG_APP_UPDATE_CHECK_APP_HASH=y
CONFIG_ESP8266_NMI_WDT=y				     |	# CONFIG_ESP8266_NMI_WDT is not set
# CONFIG_SOC_FULL_ICACHE is not set			     |	CONFIG_SOC_FULL_ICACHE=y
CONFIG_SOC_IRAM_SIZE=0xC000				     |	CONFIG_SOC_IRAM_SIZE=0x8000
CONFIG_WIFI_PPT_TASKSTACK_SIZE=5120			     |	CONFIG_WIFI_PPT_TASKSTACK_SIZE=4096
CONFIG_ESP8266_CORE_GLOBAL_DATA_LINK_IRAM=y		     <
# CONFIG_ESP8266_WIFI_AMPDU_RX_ENABLED is not set	     |	CONFIG_ESP8266_WIFI_AMPDU_RX_ENABLED=y
							     >	CONFIG_ESP8266_WIFI_RX_BA_WIN_SIZE=6
CONFIG_ESP8266_WIFI_RX_BUFFER_NUM=16			     |	CONFIG_ESP8266_WIFI_RX_BUFFER_NUM=28
CONFIG_ESP8266_WIFI_LEFT_CONTINUOUS_RX_BUFFER_NUM=16	     |	CONFIG_ESP8266_WIFI_LEFT_CONTINUOUS_RX_BUFFER_NUM=8
CONFIG_ESP8266_WIFI_RX_PKT_NUM=7			     |	CONFIG_ESP8266_WIFI_RX_PKT_NUM=16
CONFIG_ESP8266_PHY_MAX_WIFI_TX_POWER=20			     |	CONFIG_ESP8266_PHY_MAX_WIFI_TX_POWER=21
# CONFIG_ESP8266_HSPI_HIGH_THROUGHPUT is not set	     |	CONFIG_ESP8266_HSPI_HIGH_THROUGHPUT=y
CONFIG_ESP_ERR_TO_NAME_LOOKUP=y				     |	# CONFIG_ESP_ERR_TO_NAME_LOOKUP is not set
CONFIG_ESP_TIMER_TASK_STACK_SIZE=3584			     |	CONFIG_ESP_TIMER_TASK_STACK_SIZE=2048
CONFIG_ESP_TASK_WDT=y					     |	# CONFIG_ESP_TASK_WDT is not set
CONFIG_ESP_TASK_WDT_PANIC=y				     <
CONFIG_FREERTOS_GLOBAL_DATA_LINK_IRAM=y			     |	CONFIG_FREERTOS_CODE_LINK_TO_IRAM=y
# CONFIG_FREERTOS_CODE_LINK_TO_IRAM is not set		     <
# CONFIG_HEAP_DISABLE_IRAM is not set			     |	CONFIG_HEAP_DISABLE_IRAM=y
# CONFIG_LOG_DEFAULT_LEVEL_ERROR is not set		     |	CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
CONFIG_LOG_DEFAULT_LEVEL_INFO=y				     |	# CONFIG_LOG_DEFAULT_LEVEL_INFO is not set
CONFIG_LOG_DEFAULT_LEVEL=3				     |	CONFIG_LOG_DEFAULT_LEVEL=1
# CONFIG_LWIP_IRAM_OPTIMIZATION is not set		     |	CONFIG_LWIP_IRAM_OPTIMIZATION=y
CONFIG_LWIP_IP6_FRAG=y					     |	# CONFIG_LWIP_IP6_FRAG is not set
CONFIG_LWIP_ESP_GRATUITOUS_ARP=y			     |	# CONFIG_LWIP_ESP_GRATUITOUS_ARP is not set
CONFIG_LWIP_GARP_TMR_INTERVAL=60			     |	CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=32			     <
CONFIG_LWIP_NETIF_LOOPBACK=y				     |	# CONFIG_LWIP_NETIF_LOOPBACK is not set
CONFIG_LWIP_LOOPBACK_MAX_PBUFS=8			     <
CONFIG_LWIP_MAX_LISTENING_TCP=16			     |	CONFIG_LWIP_MAX_LISTENING_TCP=8
CONFIG_LWIP_TCP_MSS=1440				     |	CONFIG_LWIP_TCP_MSS=1460
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=2880			     |	CONFIG_LWIP_TCP_SND_BUF_DEFAULT=11680
CONFIG_LWIP_TCP_WND_DEFAULT=5760			     |	CONFIG_LWIP_TCP_WND_DEFAULT=10240
CONFIG_LWIP_TCP_RECVMBOX_SIZE=6				     |	CONFIG_LWIP_TCP_RECVMBOX_SIZE=16
CONFIG_LWIP_MAX_UDP_PCBS=16				     |	CONFIG_LWIP_MAX_UDP_PCBS=4
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=2048			     |	CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=4096
CONFIG_LWIP_MAX_RAW_PCBS=16				     |	CONFIG_LWIP_MAX_RAW_PCBS=4
# CONFIG_ESP_SHA is not set				     |	CONFIG_ESP_SHA=y
CONFIG_TCPIP_ADAPTER_GLOBAL_DATA_LINK_IRAM=y		     <
CONFIG_OPTIMIZATION_LEVEL_DEBUG=y			     |	# CONFIG_OPTIMIZATION_LEVEL_DEBUG is not set
# CONFIG_OPTIMIZATION_LEVEL_RELEASE is not set		     |	CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED=y		     |	# CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED is not set
# CONFIG_OPTIMIZATION_ASSERTIONS_DISABLED is not set	     |	CONFIG_OPTIMIZATION_ASSERTIONS_DISABLED=y
CONFIG_TASK_WDT=y					     |	# CONFIG_TASK_WDT is not set
CONFIG_TASK_WDT_PANIC=y					     <
CONFIG_ESP_GRATUITOUS_ARP=y				     |	# CONFIG_ESP_GRATUITOUS_ARP is not set
CONFIG_GARP_TMR_INTERVAL=60				     |	CONFIG_TCPIP_RECVMBOX_SIZE=64
CONFIG_TCPIP_RECVMBOX_SIZE=32				     <
CONFIG_TCP_MSS=1440					     |	CONFIG_TCP_MSS=1460
CONFIG_TCP_SND_BUF_DEFAULT=2880				     |	CONFIG_TCP_SND_BUF_DEFAULT=11680
CONFIG_TCP_WND_DEFAULT=5760				     |	CONFIG_TCP_WND_DEFAULT=10240
CONFIG_TCP_RECVMBOX_SIZE=6				     |	CONFIG_TCP_RECVMBOX_SIZE=16
CONFIG_TCPIP_TASK_STACK_SIZE=2048			     |	CONFIG_TCPIP_TASK_STACK_SIZE=4096
```
) 
Thx

windowsair, 2022-09-17:
For esp8266, please use the SDK that comes with the project.

gitthnx, 2022-09-17:
i think i did, with each tagged version tested to compile 
(<code>git clone --recursive <wireless-esp8266-dap> -b v0.1.3.1 && cd ESP8266_RTOS_SDK && ./install.sh && . ./export.sh && export IDF_PATH=\`pwd\` && cd..</code>)
first successful attempt generating an app .bin binary (smaller than .elf file) was with 'copying sdkconfig file hello_world/sdkconfig replacing wireless-esp8266-dap/sdkconfig, compiling done, copying original wireless-esp2866-dap/sdkconfig.backup to sdkconfig, now compiling (why? & tbd: soon :), being repeated for verification)

windowsair, 2022-09-17:
It is true that most cases are related to submoule. btw you are using an older version, so if you still have problems, try the newer version of the project.

closed 2022-09-25 by gitthnx