windowsair/wireless-esp8266-dap #42: Release的固件没有正常运行

LDEDA4587, 2022-10-10:
模块是ESP-WROOM-02, 16Mbit.使用的固件是v0.2.0,下载设置如下:
[image]
启动信息如下:
[image]
SPI MODE设置为QIO,DIO,DOUT启动信息都一样,在网络里也找不到AP.
v0.1.2版本的固件也试过,都是无法正常运行.
这是下载设置哪里有问题吗?

windowsair, 2022-10-10:
使用full固件烧录到0x0

LDEDA4587, 2022-10-10:
[image]
[image]
这样就是成功运行了吗?

windowsair, 2022-10-10:
是的，这样就烧录成功了

LDEDA4587, 2022-10-10:
非常感谢! 这样看来ESP-12F(Nodemcu)也是可以使用的.

closed 2022-10-10 by windowsair