windowsair/wireless-esp8266-dap #45: not booting

jandy123, 2022-11-07:
I have a problem running the latest firmware. It all build up fine, flashing works fine, but after reboot I cannot find any new DAP wifi device. 

Is there any way to debug this? Maybe via aerial port messages, since I have no clue if it's booting or not. 

Thanks!

jandy123, 2022-11-08:
Turns out I'm having issues with the power supply. I use an ESP-12F based on the schematics posted on the website. 
This evening will try with a better power supply and report back. 

Thanks a lot for the quick reaction!

jandy123, 2022-11-08:
Well, power supply problem fixed -- using a power bench supply, but still issues.

- Many times, after loading the modules  and attaching using ` sudo usbip attach -r 192.168.2.165 -b 1-1` (linux here), I get errors like (using dmesg): 

[24401.730441] usb 5-1: USB disconnect, device number 8
[24405.798174] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[24405.798176] vhci_hcd vhci_hcd.0: devid(65537) speed(3) speed_str(high-speed)
[24406.037306] usb 5-1: new high-speed USB device number 11 using vhci_hcd
[24406.169350] usb 5-1: SetAddress Request (11) to port 0
[24406.199523] usb 5-1: recv xbuf, 0
[24406.199773] vhci_hcd: stop threads
[24406.199777] vhci_hcd: release socket
[24406.199786] vhci_hcd: disconnect device

Obviously, OpenOCD fails to find the device with:
 
Error: 95 534 cmsis_dap.c:305 cmsis_dap_open(): unable to find a matching CMSIS-DAP device
Debug: 96 534 command.c:556 run_command(): Command 'init' failed with error code -4
User : 97 534 command.c:619 command_run_line(): 

- Sometimes, I do get the proper output:
`[24815.067472] usb 5-1: New USB device found, idVendor=c251, idProduct=f00a, bcdDevice= 1.00
[24815.067478] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[24815.067481] usb 5-1: Product: Wireless ESP CMSIS-DAP
[24815.067484] usb 5-1: Manufacturer: windowsair
[24815.067486] usb 5-1: SerialNumber: 1234
`
but then OpenOCD fails with:
`Info : 100 377 adapter.c:108 adapter_init(): clock speed 1000 kHz
Debug: 101 377 openocd.c:143 handle_init_command(): Debug Adapter init complete
Debug: 102 377 command.c:166 script_debug(): command - transport init
Debug: 103 377 transport.c:230 handle_transport_init(): handle_transport_init
Debug: 104 377 command.c:166 script_debug(): command - dap init
Debug: 105 377 arm_dap.c:107 dap_init_all(): Initializing all DAPs ...
Debug: 106 390 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 107 413 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 108 428 cmsis_dap.c:1115 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 109 455 cmsis_dap.c:1130 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 110 486 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 111 500 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 112 520 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 113 535 cmsis_dap.c:1115 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 114 576 cmsis_dap.c:1130 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 115 608 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 116 625 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 117 645 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 118 662 cmsis_dap.c:1115 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 119 694 cmsis_dap.c:1130 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 120 719 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 121 734 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 122 756 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 123 775 cmsis_dap.c:1115 cmsis_dap_swd_switch_seq(): JTAG-to-DORMANT
Debug: 124 805 cmsis_dap.c:1130 cmsis_dap_swd_switch_seq(): DORMANT-to-SWD
Debug: 125 836 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 126 853 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Debug: 127 878 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Error: 128 879 adi_v5_swd.c:342 swd_connect_single(): Error connecting DP: cannot read IDR
Debug: 129 879 command.c:556 run_command(): Command 'dap init' failed with error code -4
User : 130 879 command.c:619 command_run_line(): 
Debug: 131 879 command.c:556 run_command(): Command 'init' failed with error code -4
User : 132 879 command.c:619 command_run_line(): 
Debug: 133 879 target.c:2204 target_free_all_working_areas_restore(): freeing all working areas
Debug: 134 895 cmsis_dap.c:1125 cmsis_dap_swd_switch_seq(): SWD-to-DORMANT
Debug: 135 925 cmsis_dap.c:1135 cmsis_dap_swd_switch_seq(): DORMANT-to-JTAG
` 

Btw. I'm running OCD using:
`sudo ./src/openocd -s ./tcl -f interface/cmsis-dap.cfg -f target/stm32f1x.cfg`

after having properly connected an stm32f1xx board -- which is known to work with other STW probes.

Any idea what I could try?

Thanks!

windowsair, 2022-11-09:
It seems to be a usbip problem... I'm sorry that I haven't tested under Linux for a long time

jandy123, 2022-11-09:
I did make some progress. In my attempts to get it to work when the power supply was a problem, I did some modifications which made the setup totally fail. 

However, today, after just cloning the repo and compiling everything, I am able to get the proper `SWD DPIDR 0x0bb11477`, but then OCD stops with the error

Debug: 124 131 arm_dap.c:107 dap_init_all(): Initializing all DAPs ...
Debug: 125 136 cmsis_dap.c:1110 cmsis_dap_swd_switch_seq(): JTAG-to-SWD
Info : 126 145 adi_v5_swd.c:346 swd_connect_single(): SWD DPIDR 0x0bb11477
Debug: 127 148 cmsis_dap.c:883 cmsis_dap_swd_read_process(): SWD ack not OK @ 0 JUNK
Debug: 128 148 command.c:556 run_command(): Command 'dap init' failed with error code -4
User : 129 148 command.c:619 command_run_line(): 
Debug: 130 148 command.c:556 run_command(): Command 'init' failed with error code -4
User : 131 148 command.c:619 command_run_line(): 
Debug: 132 148 target.c:2204 target_free_all_working_areas_restore(): freeing all working areas 

As before I had to try multiple time attaching the remote device to finally get 

[19375.748518] usb 5-1: New USB device found, idVendor=c251, idProduct=f00a, bcdDevice= 1.00
[19375.748524] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19375.748527] usb 5-1: Product: Wireless ESP CMSIS-DAP
[19375.748530] usb 5-1: Manufacturer: windowsair
[19375.748532] usb 5-1: SerialNumber: 1234

After getting once the correct SWD DPIDR and failing with error -4 as above, I also see the error

[19388.165102] vhci_hcd: unlink->seqnum 584
[19388.165107] vhci_hcd: the urb (seqnum 584) was already given back

Any further attempts lead to errors in OCD. The only option is then to detach the remote USB device, reattach and start all over again to still get the -4 error above. 

So, indeed, there are issues with compatibility with USBIP, specifically the error above regarding the urb.  Anyone, any clue what that means?

windowsair, 2022-11-10:
I have to admit that this project currently has limited support for OpenOCD, and the issues you mention are ones I am currently experiencing. OpenOCD differs significantly from Keil in some of its processing. For example, in scenarios where the network fluctuates, OpenOCD will throw an error causing debugging to terminate.

windowsair, 2022-11-10:
1. For the first problem, my current approach is to use a lower rate when connecting to OpenOCD, and when the connection is established, I manually switch to a higher rate (10MHz) in console, but this can also be done using scripts.

2. These pins are irrelevant, for example, when you only use SWD, you don't need to care about pins such as TDI/TDO used by JTAG.

windowsair, 2022-11-10:
But I have to admit that the current OpenOCD experience is still terrible. I will work on fixing this in the future, perhaps by fixing some of the timing issues with OpenOCD, or perhaps by just completing a specific OpenOCD that doesn't require usbip.  😃

jandy123, 2022-11-10:
> But I have to admit that the current OpenOCD experience is still terrible. I will work on fixing this in the future, perhaps by fixing some of the timing issues with OpenOCD, 

Yes, OpenOCD seems to be very sensitive to the TCP transfer speed (point 2 in my list). Would be nice to make it more resilient., albeit at some speed cost.

> or perhaps by just completing a specific OpenOCD that doesn't require usbip. smiley

Yes, I've  been thinking about this too. Of course we have remote bitbang and the like, but performance is terrible. Some sort of TCP interface would be needed to make the CMSIS-DAP compatible with TCP, without the usbip. But then, I'm not even sure if usbip is the problem. Will test this too.

jandy123, 2022-11-10:
> 1. For the first problem, my current approach is to use a lower rate when connecting to OpenOCD, and when the connection is established, I manually switch to a higher rate (10MHz) in console, but this can also be done using scripts.

For me this miserable fails. Once I increase the speed afterwards (using telnet for instance), OOCD tries to reexamine the target but fails miserably.

Still, where in the esp code is the speed switch performed? Since you this is implied by what you say in the readme file under "Speed Strategy".

>     2. These pins are irrelevant, for example, when you only use SWD, you don't need to care about pins such as TDI/TDO used by JTAG.

Thought so, but just wanted to confirm...

windowsair, 2022-11-10:
Oh no, you may be a victim of OpenOCD. 

The CMSIS-DAP protocol contains the command to specify the speed where the speed switch will take place once we receive the command. Well, I'll take a look at OpenOCD's behavior with the logic analyzer when I have time this weekend, although I've noticed before that OpenOCD doesn't seem to meet the arm debug spec in some cases, or maybe that's just my problem.

I'm not quite sure what your intent is in focusing on these pins.  Probably you are trying to understand the effect of these pins on external devices when idle or something else?

jandy123, 2022-11-10:
> The CMSIS-DAP protocol contains the command to specify the speed where the speed switch will take place once we receive the command. Well, I'll take a look at OpenOCD's behavior with the logic [...]

Please let me know if you find something...
 
> I'm not quite sure what your intent is in focusing on these pins. Probably you are trying to understand the effect of these pins on external devices when idle or something else?

I do not understand why I need to have the serial2usb connection up and running to get a stable connection. I do not intend to connect anything more than the SWD pins for SWD debugging/flashing and JTAG pins for JTAG.

windowsair, 2022-11-10:
It' s weird, maybe it' s a power issue?

jandy123, 2022-11-10:
> It' s weird, maybe it' s a power issue?

I doubt that this is the problem. I fixed that. May still be something with the grounding/wiring, though. As I said, will test this properly this evening and get back.

jandy123, 2022-11-10:
It's fixed! I updated my findings above.

What is the flash reading/writing speed in windows with OpenOCD? How about Keil?

windowsair, 2022-11-11:
awesome!  Maybe OpenOCD is sensitive to line connections.

You seem to be measuring a pure read/write speed, but this is of little relevance to the actual speed in use, as there are effects such as TCP latency.

But in general, Keil should be faster than OpenOCD. Keil with elaphureLink should have the fastest speed.

jandy123, 2022-11-11:
@windowsair Thanks a lot for your input and very useful firmware!

Maybe the status "bug" should bee removed and instead called "getting it to work with OpenOCD & linux" (and avoiding stupid gotchas ;).

closed 2024-06-09 by windowsair