windowsair/wireless-esp8266-dap #46: ESP8266 does not boot after flashing

khorght, 2022-11-08:
Hi. Tryed both merged and separate bin files versiont 0.1.4 then 0.2.0 and 0.2.1, wrote merged at 0x0000 and separate bin files -  bootloader to 0x0000, partition_table to 0x8000 and app to 0x10000. Used "ESP download flash tool" v3.9.3 to burn binaries. Device gives to console this output
`load 0x40100000, len 5880, room 16 

tail 8

chksum 0xeb

load 0x3ffe8008, len 24, room 0 

tail 8

chksum 0x43

load 0x3ffe8020, len 3308, room 0 

tail 12

chksum 0xd4

csum 0xd4
[0;32mI (59) boot: ESP-IDF ed19d9c 2nd stage bootloader[0m

[0;32mI (59) boot: compile time 06:01:02[0m

[0;32mI (68) qio_mode: Enabling default flash chip QIO[0m

[0;32mI (68) boot: SPI Speed      : 80MHz[0m

[0;32mI (72) boot: SPI Mode       : QIO[0m

[0;32mI (78) boot: SPI Flash Size : 16MB[0m

[0;32mI (84) boot: Partition Table:[0m

[0;32mI (90) boot: ## Label            Usage          Type ST Offset   Length[0m

[0;32mI (101) boot: End of partition table[0m

[0;31mE (108) boot: No bootable app partitions in the partition table[0m

user code done`

board is custom diy ESP8266 PCB with 4Mb flash, but i've tested it with primitive arduino apps, all works fine not a hardware issue. 

Tryed to upload using ESPHome-Flasher-v1.3.0 and even using commandline Arduino esptool.py. Same result and same output to console with all versions.

khorght, 2022-11-09:
My first attempt was to burn full firmware, in first post i called it "merged", ESP8266 didn't start, with the very same output in serial port, then i've tryed to burn separate parts, then i've tryed another full version, then yet another, and all of them one more time  one after another and always i'm getting same output in terminal. Compiled Arduino blink, burned - all OK, compiled and burned WiFi AP - all OK, compiled blink in PIO - burned- OK, tryed to burn compiled blink firmware using mentioned above utilities - OK, so it's not a hardware issue or my twisted hands laying on burning utilities. Not sure how to compile your code at home because in msys2 flawor cooked by Espressif for esp8266 there are no support for esp-idf compilation based on idf.py , just regular make menuconfig -> make -> make flash, but those with hadnmade Makefile fails to compile with error - app_main() not found.

windowsair, 2022-11-09:
It's weird. I can't reproduce your error, maybe you should erase the whole chip and try flash again.

windowsair, 2022-11-09:
If you are using ESP Flash Tool, you may also need to check the "Donotchangebin" box

windowsair, 2022-11-10:
It looks like it's worse this time, even the bootloader didn't load.  Maybe you could try changing the baud rate and SPI options (DIO) when downloading.

Certain hardware problems may also cause this to happen, so replacing a chip may be the ultimate solution to the problem.😢

khorght, 2022-11-10:
Changing to DIO didn't helped. Compiled  SoftAP example from  ESP8266_RTOS_SDK, burned flawlessly, i see ESP as and AccessPoint, for me it looks like ESP-chip is OK, flash is OK, and your FW is not OK. This was my last attempt. Thanx for your answers and patience. Goog luck and buy.

closed 2022-11-10 by khorght