windowsair/wireless-esp8266-dap #47: Not booting on ESP8266 (first part is in issue #46)

khorght, 2022-11-11:
First part of getting FW to work was a fail, issue #46. Decided to make one more last try in hardcore style: compiled softAP from latest ESP8266_RTOS_SDK, opened in hex editor my compiled bootloader.img and partitions.img, compared them with relative parts in the ESP8266 wireless_esp_dap_full.bin v.0.2.1 - they are completly differs. Replaced bootloader part at 0x0000 and partitions part at 0x8000, burned this hardcooked fw.It's started, succesfully connected to the router,  successfully connected to elaphureLink and recognized by Keil.  My bootloader and partitions bin files attached to this post for further investigation. Looks like the culprit is Partitions part with multiple OTA defifnitions in it. 
[binaries.zip](https://github.com/windowsair/wireless-esp8266-dap/files/9988650/binaries.zip)

windowsair, 2022-11-11:
Thank you for your excellent work!  Are you sure there is a problem with the current "ESP8266 wireless_esp_dap_full.bin" ?  

I will check this out. Thank you for your feedback!

windowsair, 2022-11-11:
No, this does not seem to be a partition table issue. Although the partition table is different, the bootloader always prints out what is contained in the partition table, whereas the partition table is not printed out in issue #46. Notice that the SPI Mode and SPI Flash Size are not the same for both versions of the bootloader, so it is possible that this is the cause of the problem.

windowsair, 2022-11-11:
Hi khorght,  could you please try this firmware?

[firmware_esp8266.zip (1).zip](https://github.com/windowsair/wireless-esp8266-dap/files/9990865/firmware_esp8266.zip.1.zip)

closed 2024-06-09 by windowsair