windowsair/wireless-esp8266-dap #48: 关于ESP32C3的硬件复位nReset(GPIO05)

Su1kAze, 2022-11-15:
esp32c3的硬件复位似乎无效。
示波器无法看到nReset引脚被拉低。
板子用的是合宙的简约版开发板，esp32c3与被调试芯片用5根10cm的高温线连接，算上接触电阻不超过5Ω。
下载的3个固件均无法拉低nReset。
固件1：通过源码还原工程，编译并下载；
固件2：您这边提供的编译好的esp32c3固件文件夹，wireless_esp_dap_full.bin；
固件3：合宙编译好的soc
三个固件下载后，现象一致，使用elaphureLin，在keil5.35环境下，勾选Reset and Run。
被调试芯片选择stm32f103、stm32f107、gd32f407和air32f103，均可以找到被调试芯片，下载速度设为1M。
[image]
强制使用硬件复位选项，会提示RDDI-DAP Error，无法下载或擦除。
[image]
[image]
使用自动或软件复位，可以擦除下载程序，会提示RDDI-DAP Error，无法正常软复位，需要手动复位被调试芯片。
但是可以直接Debug被调试芯片，可完成擦除下载复位操作。
[image]

windowsair, 2022-11-16:
如果我没有记错的话，keil的cmsis dap的硬复位本身是有问题的，不会有任何动作

windowsair, 2022-11-16:
抱歉，没注意看是elaphureLink。

这个可能需要两边配合一下，我会在后续添加这个功能

windowsair, 2022-11-16:
另外我想问一下，这边对于复位时序有什么特殊的要求吗？

Su1kAze, 2022-11-16:
没有什么特殊需求，能跟普通带复位的daplink一样就好咯[image]

windowsair, 2022-11-20:
能否试一下以下这个版本

https://github.com/windowsair/wireless-esp8266-dap/actions/runs/3507915368

https://github.com/windowsair/elaphureLink/actions/runs/3508007303

LDEDA4587, 2022-11-20:
ESP8266已经成功复位运行，非常感谢。
[image]

Su1kAze, 2022-11-21:
esp32c3刷入wireless_esp_dap_full.bin，使用elaphureLink_Windows_x64_release下的应用程序，设置仿真器会提示未实现此功能，确认后可以看到找到了被调试芯片。
[image]
[image]
此时下载按钮是禁用的。
[image]
启动调试会提示未实现此功能。
[image]
[image]
[image]
[image]

Su1kAze, 2022-11-21:
合宙的air32f103，接在开发板上，上次测过这个设备，可以用；
刚刚换成了stm32l431 这个芯片硬复位和软复位都正常。
[image]
[image]

windowsair, 2022-11-21:
嗯嗯，目前这些没有PACK选项卡的设备暂时不支持，我会尽快支持他们

Su1kAze, 2022-11-21:
好的，我明白了，麻烦你了。[image]

windowsair, 2022-11-27:
Hello，能否试下这个版本
https://github.com/windowsair/elaphureLink/actions/runs/3557761458

LDEDA4587, 2022-11-27:
简单试了下Release版本，下载复位没问题，打上断点第一次进去Debug的时候提示一个错误，没留意看报的是什么，以为可以复现，结果后面再也没有出现过，之前的版本是不是也有这样的没太留意，等会试下。

Su1kAze, 2022-11-29:
不好意思，这两天比较忙，今天才试。
使用了新版的elaphureLink之后，调试没有pack的air32f103芯片，选择硬复位，无法进行擦除和下载，会提示：
[image]
选择自动检测复位下载程序会提示：
[image]
[image]
选择软复位只会提示自动检测复位提示的第二张图。
keil中的build output显示正常，出现了Application running ...
[image]
但选择哪种复位还是不会自动复位芯片。
选择硬复位下无法进入调试模式，会提示：
[image]
选择软复位第一次可以调试，但运行一小会后elaphureLink卡死了且无法结束进程，电脑WIFI和热点也崩了，重启后WIFI和热点恢复正常，也可能刚好是刚好PCIe转WLAN芯片崩了...。重启后软复位也无法进入调试模式了。
使用新版elaphureLink调试有pack的stm32f103，下载调试正常，keil中的build output显示正常，但软硬复位都无法自动复位芯片。
[image]
换回上一版本的elaphureLink，发现软硬复位也是无效的。上次看build output显示正常，误认为复位也正常了。

windowsair, 2022-11-29:
奇怪的回归。我后续换个国产芯片继续完善下

windowsair, 2022-12-04:
Hello @Su1kAze  , 目前我试了下air32f103cbt6，至少软件复位是没问题的。
可以试下更新最新的keil pack。
https://github.com/openLuat/luatos-soc-air32f103

windowsair, 2022-12-04:
还有就是合宙官方说air32f103的硬件复位有问题，所以改软件复位试试吧

Su1kAze, 2022-12-07:
刚才试了下gd32，现象一样诶。
编译好程序，使用软件复位，按debug下载复位调试是可以复位的。
按download下载复位就不能复位了，软复位和硬复位在这个情况下都不能复位。( ･ั﹏･ั)

Su1kAze, 2022-12-07:
下载stm32也是这样，debug可以复位，但download无法复位。

LDEDA4587, 2022-12-07:
ESP8266和ESP32C3我都做有,ESP8266完全正常,ESP32C3一共只用了3次,第一次Download提示错误,大概就是没有正确复位了,然后紧跟着又Download了两次,竟然都正常了.因为ESP32C3的那一块板别的电路还有问题没解决,目前没办法进一步确定到底是不是完全正常,只能确定正常Download过两次.另外,两块板的硬复位我都是接了的.

LDEDA4587, 2022-12-07:
当然,我用的是STM32.

windowsair, 2022-12-11:
所以是ESP32C3上的硬件复位有问题？这个我再检查一下

badboyhuan, 2023-03-21:
[image]
同样是ESP32C3上，KEIL版本为5.37
Reset :SYSRESETREQ 
[image]

勾选了Reset and Run
[image]

无法复位，但是通过Debug可以复位。
之前用stlink的时候勾选Pack的Enable也不能复位，我尝试取消勾选，如下图所示：
[image]
但是也无法复位。
烧录的板子为：STM32F429BIT6
用的上位机为elaphureLink Version：1.2.0.0——2022-08-16

badboyhuan, 2023-03-21:
似乎这个功能的修复在开发板中，发行版还没有更新，坐等更新，辛苦了，加油哦。

windowsair, 2023-11-21:
> [image] 同样是ESP32C3上，KEIL版本为5.37 Reset :SYSRESETREQ [image]
> 
> 勾选了Reset and Run [image]
> 
> 无法复位，但是通过Debug可以复位。 之前用stlink的时候勾选Pack的Enable也不能复位，我尝试取消勾选，如下图所示： [image] 但是也无法复位。 烧录的板子为：STM32F429BIT6 用的上位机为elaphureLink Version：1.2.0.0——2022-08-16

嗯，用的elaphureLink版本可以更新下看看

https://github.com/windowsair/elaphureLink/actions/runs/6614836721

mlx-core, 2024-05-10:
能不能告诉这个几个烧录的参数，我烧录半天都不对，我用esp32c3
[image]
[image]