windowsair/wireless-esp8266-dap #50: ESP32-C3FH4使用此固件WiFi信号很差.

LDEDA4587, 2022-12-13:
使用ESP32-C3FH4设计了一块板,使用板载天线,如下图
[image]
另外又购买了一块现成的ESP32-C3FH4开发板作参考,如下图
[image]
以上两块板使用官方AT固件,WiFi信号都很好,而使用wireless dap的固件信号就很差,基本上只有在不超过10厘米的范围内才能连上热点,即使连上了在十多分钟的时间至少要掉线四五次.不知为何会如此?这真的是一个很奇怪的问题.
看了下sdkconfig文件的功率配置,是20,但是我不知道这个20指的是dbm还是设定值,如果是AT固件设定功率的话,存在一个设定值与实际值的对应关系,但很显然config文件与API显然是不同的.

windowsair, 2022-12-14:
20dbm应该是默认值。有试过其他的值吗？

LDEDA4587, 2022-12-14:
如果20指的是dbm，那已经是最大值了。我还没有搞ESP32的编译环境，一直用的是Release的固件。
今天又做了个试验，应该可以确定与固件有关。试验如下：
用上面两块板中任一块刷入官方AT固件，然后用AT+CWJAP连接PC端的热点，然后用AT+CWJAP？查看ESP32状态，电脑端用arp -a查看与热点建立连接的客户端IP，结果发现，间距1米时，ESP32报告的RSSI一直稳定在-40至-52之间，而·ESP32仍然在10多分钟内掉线多达7次以上。可见，问题的根源不是信号差，而是掉线再重连，这个问题在Win10下可能表现不明显，但是在Win7下elaphurelink需要手动输入IP时却是致命问题。

LDEDA4587, 2022-12-14:
额，我搞错了，似乎正好证明了与固件无关。

LDEDA4587, 2022-12-14:
额，事件比较多，糊涂了，看错数据。纠正下，疯狂掉线重连的仍然是刷了wireless DAP固件的ESP32C3，而刷了AT固件的10多分钟内IP没变过，RSSI一直稳定在-40至-52之间，非常稳定。确实是与wireless DAP固件有关，表现就是疯狂变IP。晚上回去上图，目前不方便。

LDEDA4587, 2022-12-14:
刷了wireless DAP固件的ESP32C3,IP不停变动:
[image]
一个刷wireless DAP固件,一个刷官方AT固件,同时连上同一个热点,变动的还是刷了刷了wireless DAP固件的那个:
[image]
另外一个刷官方AT固件的RSSI一直稳定:
[image]
刚才用手机开热点,用刷了wireless DAP固件的ESP32C3连接,距离2米,15分钟内IP都没有变化,莫非又是Win7的热点的问题?但是刷了wireless DAP固件的ESP8266连Win7的热点很稳定啊,搞不明白了.

windowsair, 2022-12-15:
抱歉，最近太忙了，没有及时回复

关于这个问题，暂时没有什么好的解决方案。因为项目本身关于WIFI连接的部分就是用的乐鑫官方提供的demo。如果可以的话，还请测试下官方SDK中的简单的demo是否也存在这样的问题

LDEDA4587, 2022-12-16:
嗯,理解的.
今天在Win10下做了测试,先说结论:Win10也存在同样的问题,连接不畅,经常各种报错,或者点了Download后要过5秒以上才有反应,而且下载速度比ESP8266慢得多,还不稳定.偶尔也会有非常流畅的,仅仅是偶尔.下载一旦开始,就可以完整地完成并且复位,不存在其它问题.
IP是静态的,所以不会变:
[image]
[image]
下载完成:
[image]
后面我会根据你的建议跟进看看,如果还有更好的建议,也请赐教.

windowsair, 2022-12-16:
用wireshark抓包看看，看丢包是不是比较多

LDEDA4587, 2022-12-16:
完整的板没带回来,手上只有那个开发板.用一直ping的方法,发现很奇怪的现象:如果一直ping,则通讯基本很好,即使拉开几米的距离,仅偶尔会有超时,但还会继续,IP也不会变,一旦停止ping数秒之后,IP就变了.
[image]

LDEDA4587, 2022-12-16:
1k字节
[image]

LDEDA4587, 2023-03-24:
已经可以确定不是丢包的问题.
[image]
上图所示信息,每间隔大致相同时间就会重复出现,一直循环,IP不停的变.
[image]
elaphureLink连接后亦是如此.

LDEDA4587, 2024-10-12:
> 可以试试在congfig文件中配置flash大小为芯片的最大值，我也遇到了相似的问题，现象和解决过程如下： 使用的是esp32模组，flash大小为4m.
> 
> 在config文件中，原始flash配置为： CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y CONFIG_ESPTOOLPY_FLASHSIZE="2MB" 烧录测试时，距离路由器三米左右，路由器显示模块以连上wifi且会概率性丢失连接，使用ping指令间歇性可以ping通，登录ota网站：[http://corsacota.surge.sh/?address=dap.local:3241无法加载。](http://corsacota.surge.sh/?address=dap.local:3241%E6%97%A0%E6%B3%95%E5%8A%A0%E8%BD%BD%E3%80%82) 将模块与路由器的距离缩短到30cm以内，则可ping通，且ota网站可登录。
> 
> 将config改为： CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y CONFIG_ESPTOOLPY_FLASHSIZE="4MB" 再进行测试，则在3m左右的距离可实现正常功能。

你这个方法我没有测试，我使用另外一位网友基于此源代码为合宇ESP32C3开发板编译的USB烧录固件，然后问题就解决了，估计问题同源。

LDEDA4587, 2024-10-17:
> 可以试试在congfig文件中配置flash大小为芯片的最大值，我也遇到了相似的问题，现象和解决过程如下： 使用的是esp32模组，flash大小为4m.
>
> 在config文件中，原始flash配置为： CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y CONFIG_ESPTOOLPY_FLASHSIZE="2MB" [...]

请问你使用本地编译还是云编译？本来想按你说的方法尝试编译一下，没想到现在云编译报错，如图
[image]

LDEDA4587, 2024-12-11:
> 可以试试在congfig文件中配置flash大小为芯片的最大值，我也遇到了类似的问题，现象和解决过程如下： 使用的是esp32模组，flash大小为4m。
> 
> 在config文件中，原始flash配置为： CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y CONFIG_ESPTOOLPY_FLASHSIZE=“2MB” 烧录测试时，距离路由器三米左右，路由器显示模块以连上wifi且会概率性丢失连接，使用ping指令间歇性可以ping通，登录ota网站：[http://corsacota.surge.sh/?address=dap.local:3241 无法加载。](http://corsacota.surge.sh/?address=dap.local:3241%E6%97%A0%E6%B3%95%E5%8A%A0%E8%BD%BD%E3%80%82) 将模块与路由器的距离缩短到30cm以内，则可ping通，且ota网站可登录。
> 
> 将config改为： CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y CONFIG_ESPTOOLPY_FLASHSIZE=“4MB” 再进行测试，则在3m左右的距离可实现正常功能。

使用你说的方法确实可以100%解决问题，非常感谢！几天实际测试下来稳定使用，虽然下载速度比有线高速版稍慢，但是反应速度(指从点击下载到走进度条的反应)与有线基本无异，此前连接不好的时候反应总会慢一两秒，下载中途也有顿挫。至于为什么这样改就OK了，本人能力有限，没法深究。