windowsair/wireless-esp8266-dap #63: 以后会出支持esp32s3的版本吗？

dianjixz, 2023-11-10:


windowsair, 2023-11-10:
S3相比其他系列有什么优势吗？

dianjixz, 2023-11-11:
我有一个M5的ATOMS3,小小的带屏幕，移植上 daplink 兼职完美！

windowsair, 2023-11-11:
好的，我会在下一个大版本把S3支持加上

Whitolf, 2023-11-14:
我也期待可以适配esp32s3呀

AILIFE4798, 2024-01-16:
can s3 support wired + wifi connection
because it have hardware usb
so why not have both

windowsair, 2024-04-19:
#72 This pr add support for esp32s3

Whitolf, 2024-04-29:
我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题

windowsair, 2024-04-29:
> 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题

用什么进行下载的？ Keil 、 OpenOCD 还是什么？

Whitolf, 2024-04-29:
> > 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题
> 
> 用什么进行下载的？ Keil 、 OpenOCD 还是什么？

openocd

Whitolf, 2024-04-29:
我代码里面加了打印调试的，发现问题出在spi收不到数据，示波器查看clk和sdi有数据

windowsair, 2024-04-29:
> > > 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题
> > 
> > 
> > 用什么进行下载的？ Keil 、 OpenOCD 还是什么？
> 
> openocd

这个是共性问题，不止 esp32s3 ，别的芯片也会有。具体的一直没时间修复

Whitolf, 2024-04-29:
> > > > 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题
> > > 
> > > 
> > > 用什么进行下载的？ Keil 、 OpenOCD 还是什么？
> > 
> > 
> > OpenOCD的
> 
> 这个是共性问题，不止 esp32s3 ，别的芯片也会有。具体的一直没时间修复

目前用了一个比较笨的办法解决了，检测到 DAP_HostStatus 状态改变时，MCU模拟一遍切换下载速度

windowsair, 2024-06-09:
> > > > > 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题
> > > > 
> > > > 
> > > > 用什么进行下载的？ Keil 、 OpenOCD 还是什么？
> > > 
> > > 
> > > OpenOCD的
> > 
> > 
> > 这个是共性问题，不止 esp32s3 ，别的芯片也会有。具体的一直没时间修复
> 
> 目前用了一个比较笨的办法解决了，检测到 DAP_HostStatus 状态改变时，MCU模拟一遍切换下载速度

这个应该修复了，可以试下最新的版本。

Whitolf, 2024-06-14:
> > > > > > 我试了下esp32s3的版本，第一次下载的时候读不到芯片ID，要切换一下速度，切换成低速（小于10M，代码里面对应非SPI模式），然后再切换成高速（SPI模式）才可以下载，不知道哪有问题
> > > > > 
> > > > > 
> > > > > 用什么进行下载的？ Keil 、 OpenOCD 还是什么？
> > > > 
> > > > 
> > > > OpenOCD的
> > > 
> > > 
> > > 这个是共性问题，不止 esp32s3 ，别的芯片也会有。具体的一直没时间修复
> > 
> > 
> > 目前用了一个比较笨的办法解决了，检测到 DAP_HostStatus 状态改变时，MCU模拟一遍切换下载速度
> 
> 这个应该修复了，可以试下最新的版本。

测试了一下，确实没问题了

closed 2024-06-15 by windowsair