windowsair/wireless-esp8266-dap #64: esp32 ，软件复位不起作用

sq113, 2023-11-16:
[image]
[image]

windowsair, 2023-11-16:
能否提供一个最小的复现工程？

sq113, 2023-11-16:
硬件esp32 stm32f103
固件  https://github.com/windowsair/wireless-esp8266-dap/releases 预编译固件
软件keil5.36  elaphureLink 1.2.0稳定版

在换成keil5.38 和 elaphureLink 最新开发版 后也是一样但是烧录结束会额外出现窗口
[image]
[image]
[image]

windowsair, 2023-11-16:
看起来像是启用了硬件复位？

方便提供下最小复现的Keil工程吗

windowsair, 2023-11-16:
另外，请勾选下SWJ选项再测试下

sq113, 2023-11-16:
[led.zip](https://github.com/windowsair/wireless-esp8266-dap/files/13373339/led.zip)
还是不行

windowsair, 2023-11-16:
好的，我晚点测试下

windowsair, 2023-11-16:
具体用的是哪款103芯片？103r8t6？

sq113, 2023-11-17:
stm32f103rct6

windowsair, 2023-11-21:
您好，建议您使用下面这个版本的elaphureLink再试下，在这个版本上我这里没有复现您的问题。
https://github.com/windowsair/elaphureLink/actions/runs/6614836721

记得重新安装elaphureLink驱动

closed 2024-06-09 by windowsair