windowsair/wireless-esp8266-dap #68: pyocd does not detect debug probe

AILIFE4798, 2024-01-22:
esp8266 version worked just fine with usbip on keil i can upload and debug
but i want to erase flash to unlock readout protection with pyocd but it does not detect the debug probe
[image]

windowsair, 2024-01-23:
Hi, @AILIFE4798 ,

I haven't tested pyocd. Can the device be detected if you use openocd?

AILIFE4798, 2024-01-23:
idk how to use openocd , and it does not support the mm32spin05pf mcu i need it for
the device is detected in device manager correctly like in the readme
[image]

windowsair, 2024-01-23:
If possible, could you provide more information about pyocd, such as the version you're using and related commands?

AILIFE4798, 2024-01-23:
i installed the latest pyocd with pip
just install python first (ofcourse) then use `pip install pyocd`(vpn needed)
then to check the available debug probe use `pyocd list -p` it should show the probes like this
[image]

windowsair, 2024-01-25:
I have reproduced the problem.  It may take some time to adapt pyocd :)

windowsair, 2024-01-26:
try this firmware

https://github.com/windowsair/wireless-esp8266-dap/actions/runs/7656753399

AILIFE4798, 2024-01-26:
need change wifi credentials
ssid:AILIFE_SmartHome_2.4G
password:43214321
because i cannot build my self

AILIFE4798, 2024-01-26:
ota also does not work

windowsair, 2024-01-26:
Some of pyocd's usb operations don't seem to be well adapted.  It may take a while to fix this.

AILIFE4798, 2024-01-26:
ok

windowsair, 2024-02-16:
Hello, you can try the latest code in the master branch. I've done a simple test using pyocd :)

AILIFE4798, 2024-03-08:
i have tried to use latest firmware
it can detect the probe now
but cannot erase or flash
when a target is connected correctly it says no ack received
and when i didnt connect target it says unexpected ack value 0
[image]
and there is suvior bug idk in where
cuz when i first tried this i connected the swd line wrong and when i run erase command it hangs for 30 second and entire windows os crashed with blue screen
it hapenned 2 times
 but it is not consistant i cannot reproduce it

windowsair, 2024-03-10:
Sometimes usbip can cause BSOD. For your device, is it always unable to erase?

AILIFE4798, 2024-03-10:
yes always hangs for a long time then says no ack received

AILIFE4798, 2024-03-10:
usb ip on windows is really not easy maybe better way is to upload hex or bin file via a website and itll get flashed to device
or maybe gdb server via wifi

windowsair, 2024-03-10:
Well, I don't have too many chips to test. Looks like you have mm32 and gd32 boards? If you use the gd32, will it work?

AILIFE4798, 2024-03-10:
it is doing something this time
[image]

AILIFE4798, 2024-03-10:
also tested on stm32f103 the single most common and most cloned chip
it did not work either
[image]

windowsair, 2024-03-11:
It's weird. looks like the behavior is the same between the different chips. I tested it on stm32f103rc and it work well :( 
I'll try c8 later.

AILIFE4798, 2024-03-11:
dont have a f103rc now
or actually i do have one but it is locked and i want to keep it

windowsair, 2024-03-11:
okay i found some problem. we should use 10MHz speed when we use esp8266.
```bash
pyocd erase -t stm32f103c8 --chip --frequency=10000000
```

closed 2024-06-09 by windowsair