windowsair/wireless-esp8266-dap #70: flash_parts: partition 0 invalid magic number 0x3333

reed-dan, 2024-01-27:
I flash wireless_esp_dap_full.bin from [V0.2.1](https://github.com/windowsair/wireless-esp8266-dap/releases/download/v0.2.1/firmware_esp8266.zip.zip) to my nodemcu using nodemcu firmware programmer. It doesn't connect to the wifi network , and when I monitor baud 74880 I see "partition 0 invalid"

'''
[0;32mI (43) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (50) boot: SPI Speed      : 40MHz[0m
[0;32mI (56) boot: SPI Mode       : QIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;31mE (68) flash_parts: partition 0 invalid magic number 0x3333[0m
[0;31mE (78) boot: Failed to verify partition table[0m
[0;31mE (86) boot: load partition table error![0m
user code done
'''

windowsair, 2024-01-27:
How did you download firmware? Please provide more information

reed-dan, 2024-01-27:
I downloaded the firmware from https://github.com/windowsair/wireless-esp8266-dap/releases/download/v0.2.1/firmware_esp8266.zip.zip. As far as flashing I flashed to offset 0x00000.

windowsair, 2024-01-27:
what about use esp flash download tool and select `DoNotChgBin` ?

[image]

reed-dan, 2024-01-27:
I just tried the esp8266 download tool, with the same result.

reed-dan, 2024-01-27:
think I am making some progress

"When I changed the QIO of SPI Mode to DIO, the problem was solved."

https://github.com/espressif/ESP8266_RTOS_SDK/issues/929

now I get
�
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c

windowsair, 2024-01-27:
looks like partition error. try to erase full chip before download

reed-dan, 2024-01-27:
I broke out esptool.py
----------------------------------
erase
----------------------------------
C:\Users\CR\Downloads\firmware_esp8266_og>py -m esptool --chip esp8266 erase_flash
esptool.py v4.7.0
Found 1 serial ports
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:08:f8:a5
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.5s
Hard resetting via RTS pin...
------------------------
Flash
-------------------------
C:\Users\CR\Downloads\firmware_esp8266_og>py -m esptool --chip esp8266 --port com3 write_flash --flash_mode dio --flash_size detect 0x0 wireless_esp_dap_full.bin
esptool.py v4.7.0
Serial port com3
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.
.
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 60:01:94:08:f8:a5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00076fff...
Flash params set to 0x024f
Compressed 486912 bytes to 301874...
Wrote 486912 bytes (301874 compressed) at 0x00000000 in 26.8 seconds (effective 145.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
----------------------
Result
----------------------
[0;32mI (43) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (50) boot: SPI Speed      : 80MHz[0m
[0;32mI (56) boot: SPI Mode       : QIO[0m
[0;32mI (62) boot: SPI Flash Size : 4MB[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (85) boot: End of partition table[0m
[0;31mE (92) boot: No bootable app partitions in the partition table[0m
user code done

windowsair, 2024-01-28:
I cannot reproduce your problem on my own nodemcu. If your nodemcu still doesn't work, try compiling the firmware and changing the flash size and SPI mode. And then using idf.py to flash instead of esptool.

```
python ./idf.py -p /dev/ttyS5 flash
```

If this still doesn't work, try the hello world example program in the esp8266 SDK and see if it works. If it still doesn't work, consider whether it's a hardware  issue.

reed-dan, 2024-01-28:
Hello World compiled and worked as expected following this [guide](https://github.com/espressif/ESP8266_RTOS_SDK), with SPI DIO otherwise I get "�[0;31mE (68) flash_parts: partition 0 invalid magic number 0x3333�[0m" error

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7536, room 16
tail 0
chksum 0x53
load 0x3ffe8408, len 24, room 8
tail 0
chksum 0xda
load 0x3ffe8420, len 3468, room 8
tail 4
chksum 0xfc
csum 0xfc
I (80) boot: ESP-IDF v3.4-96-gc4634015 2nd stage bootloader
I (80) boot: compile time 16:50:46
I (80) boot: SPI Speed      : 40MHz
I (93) boot: SPI Mode       : DIO
I (105) boot: SPI Flash Size : 2MB
I (118) boot: Partition Table:
I (129) boot: ## Label            Usage          Type ST Offset   Length
I (152) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (175) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (198) boot:  2 factory          factory app      00 00 00010000 000f0000
I (222) boot: End of partition table
I (235) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x1cc5c (117852) map
0x40210010: _stext at ??:?

I (336) esp_image: segment 1: paddr=0x0002cc74 vaddr=0x4022cc6c size=0x06d64 ( 28004) map
I (355) esp_image: segment 2: paddr=0x000339e0 vaddr=0x3ffe8000 size=0x00544 (  1348) load
I (359) esp_image: segment 3: paddr=0x00033f2c vaddr=0x40100000 size=0x00080 (   128) load
I (386) esp_image: segment 4: paddr=0x00033fb4 vaddr=0x40100080 size=0x0512c ( 20780) load
I (425) boot: Loaded app from partition at offset 0x10000
Hello world!
This is ESP8266 chip with 1 CPU cores, WiFi, silicon revision 1, 2MB external flash
Restarting in 10 seconds...

```

When I tried to compile wireless esp8266 dap with idf.py

```
$ idf.py build
'cmake' must be available on the PATH to use C:/msys32/home/CR/esp/esp-idf/tools
/idf.py

```
Same with idf.py menuconfig

So I added a MakeFile and ran make menuconfig which worked

````
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#

PROJECT_NAME := wireless-esp8266-dap

include $(IDF_PATH)/make/project.mk

````

Then to flashing
make flash
````
Compiler version: 8.4.0
Python requirements from C:/msys32/home/CR/esp/esp-idf/requirements.txt are satisfied.
App "wireless-esp8266-dap" version: v0.2.1-5-gcc8e819-dirty
CC build/app_update/esp_app_desc.o
AR build/app_update/libapp_update.a
Generating esp8266.project.ld
LD build/wireless-esp8266-dap.elf
c:/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:/msys32/home/CR/esp/wireless-esp8266-dap/build/esp8266\libesp8266.a(startup.o):(.literal.user_init_entry+0x1c): undefined reference to `app_main'
c:/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:/msys32/home/CR/esp/wireless-esp8266-dap/build/esp8266\libesp8266.a(startup.o): in function `user_init_entry':
C:/msys32/home/CR/esp/esp-idf/components/esp8266/source/startup.c:86: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [/home/CR/esp/esp-idf/make/project.mk:510: /home/CR/esp/wireless-esp8266-dap/build/wireless-esp8266-dap.elf] Error 1

````

windowsair, 2024-01-29:
Build wireless-esp8266-dap with these config:
```
I (80) boot: SPI Speed      : 40MHz
I (93) boot: SPI Mode       : DIO
I (105) boot: SPI Flash Size : 2MB
```

Check project README.md to see how to build.

AILIFE4798, 2024-01-29:
i am using esphome flasher software
it is very convinient only need to select serial port and the bin file
but i think i had some weird issues

With esp8266, both full and app can be flashed
App one boot with error: Bootmode:3,7 load 0x40210010,len348148,room16 (like reed-dan has encountered)
Full one boot up normally and connect to wifi

But with esp32
Full one cannot be flashed
esphome flasher Error: the firmware binary is invalid(magic byte =FF, should be E9)
And app one boot and connect to wifi normal...

so what should i use and whats the logic behind it

windowsair, 2024-01-31:
This is not an official download tool. I don't know anything about it.

reed-dan, 2024-01-31:
I ordered some esp32's, and they work without issue.

closed 2024-01-31 by reed-dan