windowsair/wireless-esp8266-dap #71: 用iar显示Probe not found

2654967934, 2024-02-27:
本来用iar昨天能连上，但是因为弹窗显示reset什么的提示，用别的烧录器也提示这个，可能开发板有问题。
今天换了一个开发板，还是用iar，但是就弹窗显示Probe not found了。用keil能正常使用。
[image]
想问iar下能正常使用的吗

2654967934, 2024-02-27:
试了下用IAR9.32.1版本的可以用
IAR8.30.1会显示Probe not found
IAR8.32.4会显示Reading CPU status failed
试了一下手里另一个有线正常的CMSISDAP，firmware version显示的也是V2.1，但是低版本的iar也能正常使用，不知道为什么

windowsair, 2024-02-28:
没有试过iar，现在用的固件版本是哪个？

2654967934, 2024-02-28:
> 没有试过iar，现在用的固件版本是哪个？

我用的就是releases里最新的

windowsair, 2024-02-28:
试试看主线分支里面最新的代码

2654967934, 2024-02-28:
> 试试看主线分支里面最新的代码

试了下情况一样

windowsair, 2024-03-08:
我看了下 IAR8.30.1 的行为，感觉 IAR 在这个版本尝试使用的是 CMSIS-DAP v1，v1版本是不支持的。

windowsair, 2024-03-08:
找到了 8.32.4 版本的 release note:
```
CMSIS-DAP v2
The debugger now supports the latest CMSIS-DAP standard with WinUSB transfer mode resulting in faster debug experiences when using CMSIS-DAP v2 compliant boards.
```

看起来这是第一个支持 v2 的版本

sblpp, 2024-04-13:
大佬，能把elaphureLink 适配到IAR吗？多谢多谢

windowsair, 2024-04-17:
> 多谢

IAR 的支持不一定能搞得到，只能先问下

closed 2024-12-24 by windowsair