windowsair/wireless-esp8266-dap #80: IDR read error

ttwards, 2024-11-03:
When I try to connect via openocd to a STM32F407IGH6, openocd reported `Error: Error connecting DP: cannot read IDR`
Here is the log from openocd
```
(base) ttwards@MiWiFi-RC06-srv [~/esp/wireless-esp8266-dap] git:(master) ✗ ➜  openocd -f ./dap.cfg
Open On-Chip Debugger 0.11.0+dev-00728-gb6f95a16c (2024-05-29-12:13)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0xc251:0xf00a, serial=1234
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.0
Info : CMSIS-DAP: Serial# = 1234
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 1 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 2000 kHz
Error: Error connecting DP: cannot read IDR
```
Here is log from ESP32(not complete)
```
SYSTEM EVENT STA GOT IP : 192.168.31.136
Connected to AP
I (6877) server_common: mDNS hostname set to: [dap]
I (6881) server_common: mDNS instance name set to: [DAP mDNS]
I (6888) corsacOTA: start corsacOTA thread...
Socket created
Socket binded
Socket listening
I (6897) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
```
Here is the data from logic analyzer:
[image]
### You can see the first HIGH level, the time gap is 6.53 us
Here is data from logic analyzer of ST-LINK-v2 at the same bitrate
[image]
You can also compare the first two signals
```
#CMSIS_DAP
Line Reset 56 bits
Line Reset 59 bits
```
```
#STLINK_v2
Line Reset 53 bits
Line Reset 56 bits
```
I'm so confused, seems when they work at both 2000kHz and sending `Request Debugport READ IDCODE`, ESP32-DAP has SWCLK at 2380 kHz, STLINK_v2 has SWCLK at 2000 kHz
I've tried 1000 kHz, they are similar

windowsair, 2024-11-04:
Try to set 1MHz or 10MHz clock before IDR read

ttwards, 2024-11-04:
```
(esp) ttwards@MiWiFi-RC06-srv [~/esp/wireless-esp8266-dap] git:(master) ✗ ➜  openocd -f ./dap.cfg                                                     [15:44:37]
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.1.0
Info : CMSIS-DAP: Serial# = 1234
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 1 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 10000 kHz
```
Setting it to 10MHz caused openocd to be stuck here
Setting it to 1MHz the problem is the same as 2MHz...

ttwards, 2024-11-04:
I thought it may be related to the version of IDF, but compiling fails with v5.3.1. Are there any plans to fix the issue with newer IDF?

windowsair, 2024-11-04:
I can not reproduce it. Check your hardware connection

ttwards, 2024-11-04:
Do models of developement boards matter? I'm currently using SEEED Xiao ESP32C3. I'll try the pre-compiled firmware later.

windowsair, 2024-12-24:
Try latest version

https://github.com/windowsair/wireless-esp8266-dap/actions/runs/12483498765

closed 2024-12-24 by ttwards

windowsair, 2024-12-24:
Great. Out of curiosity, what clock rate are you using now?

ttwards, 2024-12-24:
it works great at 2000 and 10000