windowsair/wireless-esp8266-dap #81: No discover esp8266 DAP-CMSIS

novval, 2024-11-12:
For me, esp8266 is not defined as esp8266 DAP-CMSIS, but as shown below:

D:\Downloads\usbip-win-0.3.6-dev>usbip list -r 192.168.1.199
Exportable USB devices
======================
 - 192.168.1.199
        1-1: Keil Software, Inc. : unknown product (c251:f00a)
           : /sys/devices/pci0000:00/0000:00:01.2/usb1/1-1
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/00/00)

D:\Downloads\usbip-win-0.3.6-dev>

D:\Downloads\usbip>usbip -l 192.168.1.199
- 192.168.1.199
     1-1: Keil Software, Inc. : unknown product (c251:f00a)
        : /sys/devices/pci0000:00/0000:00:01.2/usb1/1-1
        : (Defined at Interface level) (00/00/00)
        :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/00/00)

windowsair, 2024-11-13:
`c251:f00a` is PID:VID. Just connect to it.

novval, 2024-11-13:
I'm working on Windows 7.
On the latest versions of usbip, the usbip-win VHCI(ude) driver does not load. It outputs "Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)"
On older versions it is like this:
```
D:\Downloads\usbip>usbip.exe -D -a 192.168.1.199 -b 1-1
usbip dbg: usbip_network.c: 223 (tcp_connect ) trying 192.168.1.199 port 3240

usbip dbg: usbip_network.c: 243 (tcp_connect ) connected to 192.168.1.199:3240
usbip dbg: usbip_windows.c: 767 (query_interface0) exportable 1 devices
usbip dbg: usbip_windows.c: 784 (query_interface0)      1-1: Keil Software, Inc. : unknown product (c251:f00a)
usbip dbg: usbip_windows.c: 785 (query_interface0)         : /sys/devices/pci0000:00/0000:00:01.2/usb1/1-1
usbip dbg: usbip_windows.c: 786 (query_interface0)         : (Defined at Interface level) (00/00/00)
usbip dbg: usbip_windows.c: 806 (query_interface0)         :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/00/00)
usbip dbg: usbip_windows.c: 809 (query_interface0)
usbip dbg: usbip_network.c: 223 (tcp_connect ) trying 192.168.1.199 port 3240

usbip dbg: usbip_network.c: 243 (tcp_connect ) connected to 192.168.1.199:3240
usbip err: usbip_windows.c:  62 (usbip_vbus_dev_node_name) usbvbus interface is not registered

usbip err: usbip_windows.c: 654 (import_device) open vbus driver
usbip err: usbip_windows.c: 840 (attach_device) query

D:\Downloads\usbip>usbip.exe -D -a 192.168.1.199 -b 1-1
usbip dbg: usbip_network.c: 223 (tcp_connect ) trying 192.168.1.199 port 3240

usbip dbg: usbip_network.c: 243 (tcp_connect ) connected to 192.168.1.199:3240
usbip dbg: usbip_windows.c: 767 (query_interface0) exportable 1 devices
usbip dbg: usbip_windows.c: 784 (query_interface0)      1-1: Keil Software, Inc. : unknown product (c251:f00a)
usbip dbg: usbip_windows.c: 785 (query_interface0)         : /sys/devices/pci0000:00/0000:00:01.2/usb1/1-1
usbip dbg: usbip_windows.c: 786 (query_interface0)         : (Defined at Interface level) (00/00/00)
usbip dbg: usbip_windows.c: 806 (query_interface0)         :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/00/00)
usbip dbg: usbip_windows.c: 809 (query_interface0)
usbip dbg: usbip_network.c: 223 (tcp_connect ) trying 192.168.1.199 port 3240

usbip dbg: usbip_network.c: 243 (tcp_connect ) connected to 192.168.1.199:3240
usbip dbg: usbip_windows.c: 665 (import_device) call from attch here

usbip dbg: usbip_windows.c: 667 (import_device) return from attch here

usbip dbg: usbip_windows.c: 674 (import_device) devfd:0041FA88

new usb device attached to usbvbus port 1
```
The device "USB Device Over IP" appears with the error:
"The drivers for this device are not installed. (Code 28)
There is no driver selected for the device information set or element. To find a driver for this device, click Update Driver."

I am trying to build usbip on my Windows 7 machine, but so far without success (((

windowsair, 2024-11-15:
As far as I know, the kernel driver for usbip relies on windows10 :)

novval, 2024-11-15:
I try old versions, but also without success ((
Maybe the problem is specifically with my windows

B83C, 2025-06-26:
but it doesn't work when turned on 😆, only when it is disabled will it work on my machine 🤦