windowsair/wireless-esp8266-dap #82: ESP32C3版本无法下载/调试STM32H743VI

LDEDA4587, 2024-11-29:
使用elaphureLink 1.3.0，ESP32C3固件11月25日使用Actions生成，可以正常下载/调试STM32G031,STM32G070，但是对于STM32H743VI，可以在Keil正确识别，却无法下载/调试，提示如下图：
[image]
[image]
[image]
可以确定不是任何硬件问题，因为目标MCU STM32H743VI使用DAPLink和STLink都可以正常下载调试，同时此无线DAP ESP32C3版本我做了两个，两个都是对STM32G031,STM32G070使用正常，对STM32H743VI都提示同样的错误。

LDEDA4587, 2024-11-29:
补充一个在Keil中正确识别STM32H743VI的截图：
[image]

windowsair, 2024-11-29:
H743还是H750？

LDEDA4587, 2024-11-29:
确定是H743VIT6

LDEDA4587, 2024-12-10:
在Keil的Debug->Settings->Pack中，把Enable的勾选去掉，如图：
[image]
然后可以下载程序了，但是下载完之后不会自动复位运行了，并提示：
[image]
附带上Flash Download的设置：
[image]
电路上，硬件复位我也是加上了的，但是还没有100%去确认硬件复位电路是否有效，因为此前用别的MCU一直正常使用。

windowsair, 2024-12-15:
感觉是时钟的问题

LDEDA4587, 2024-12-16:
> 感觉是时钟的问题

哪一方面的时钟？给点提示。尝试把SWD时钟改为2MHz，只要把Enable勾选上，一样的结果，只要一点击Download，立马弹出上面所说的错误提示。如果不勾选Enable，甚至可以进入Debug，点击Run然后进行单步运行，勾选Enable后就不可以进入Debug。

windowsair, 2024-12-17:
SWD 时钟。感觉时序不太好，我找个时间修一下

LDEDA4587, 2024-12-25:
> 试一下最新版本的？把速率调整到 10M 以下
> 
> https://github.com/windowsair/wireless-esp8266-dap/actions/runs/12483498765

我用的是ESP32-C3FH4，需要把sdkconfig.defaults.esp32c3中的：
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
更改为：
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
才可以，否则网络会反复掉线，关于此问题此前在这里讨论过：https://github.com/windowsair/wireless-esp8266-dap/issues/50
我看你那里都是默认设置，麻烦把config也更改一下，否则我无法测试，非常感谢！

LDEDA4587, 2024-12-25:
我更新了一下，正在编译

LDEDA4587, 2024-12-25:
还是完全一样的结果，一点击Download马上就跳出错误提示，错误提示也没有任何变化，速率为1MHz，特意看了下固件版本，从2.0.0变成了2.1.0，说明我应该是更新成功了。相同的，在Debug->Settings->Pack中，把Enable的勾选去掉，也是可以下载，只是无法自动复位运行。
[image]

LDEDA4587, 2024-12-25:
补充截图
[image]

windowsair, 2024-12-25:
好的，那就是另外一个问题，有时间我看看

ICEG-F, 2025-01-24:
> 好的，那就是另外一个问题，有时间我看看

ESP32C3 0.3.0版本 + elaphureLink 1.4.0 + STM32F105

[image]

[image]
选中enable，可以下载，但是不能自动运行，
取消enable，下载不稳定，会有弹窗

ICEG-F, 2025-01-24:
> 好的，那就是另外一个问题，有时间我看看

选中enable，仿真正常，停止仿真后，程序会继续运行。