windowsair/wireless-esp8266-dap #83: 8266中LED电路、Uart使用、VScode中EIDE使用等问题

RaynorJoel, 2025-02-12:
用的8266自己画的电路板。
1.GPIO15的LED电路有问题，因为LED本身有结电容，按说明里的电路上电GPIO15是高电平，进SD启动了，改成LED并5.1k电阻上电能正常启动。
2.Uart一直没成功，配置里最后面加了CONFIG_USE_WEBSOCKET_DAP=y，重新编译的，TCP连接连不上。
3.VScode中用EIDE扩展，能支持Shell命令，如何使用本项目呢？

windowsair, 2025-02-12:
> 用的8266自己画的电路板。 1.GPIO15的LED电路有问题，因为LED本身有结电容，按说明里的电路上电GPIO15是高电平，进SD启动了，改成LED并5.1k电阻上电能正常启动。

没有明白是什么问题，电位反向了？

 > 2.Uart一直没成功，配置里最后面加了CONFIG_USE_WEBSOCKET_DAP，重新编译的，TCP连接连不上。 

`CONFIG_USE_WEBSOCKET_DAP` 选项和  UART 透传没有关系。

> 3.VScode中用EIDE扩展，能支持Shell命令，如何使用本项目呢？

结合 OpenOCD 和 pyOCD 使用。

RaynorJoel, 2025-02-12:
1.esp8266上电和复位时都会给GPIO15弱正拉，LED会微微亮，但是LED有结电容，此时GPIO15就是高电平，结果boot会进SD，串口打印boot mode:(7,7)，用模块GPIO15是悬空的，不接LED看不出来，我自己画的板接了LED，结果烧录模式都进不去。。。
2.哦，我看错行了，应该是改wifi_configuration.h吧。。。
3.OpenOCD 和 pyOCD必须得用usbip-win是吧？不能用elaphureLink了？

windowsair, 2025-02-12:
> 1.esp8266上电和复位时都会给GPIO15弱正拉，LED会微微亮，但是LED有结电容，此时GPIO15就是高电平，结果boot会进SD，串口打印boot mode:(7,7)，用模块GPIO15是悬空的，不接LED看不出来，我自己画的板接了LED，结果烧录模式都进不去。。。

这个确实是我没有考虑到，再接一个下拉电阻应该就可以了

> 3.OpenOCD 和 pyOCD必须得用usbip-win是吧？不能用elaphureLink了？

目前需要 usbip 。下一个大版本准备直接支持 openocd

RaynorJoel, 2025-02-15:
2.Uart仍然没成功，wifi_configuration.h里改了#define USE_UART_BRIDGE      1，sdkconfig.defaults.esp8266里改了CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y，CONFIG_ESPTOOLPY_FLASHSIZE="4MB"，CONFIG_SPI_FLASH_SIZE=0x4000000，上电连接好WiFi后报错，但此时我还没连接TCP发波特率数值呢，它就已经收到超长数据了。。。
Connected to AP
Socket created
Socket binded
Socket listening
E (6567) uart: uart_driver_install(927): uart rx buffer length error(>128)

打开TCP发送任意数据时串口打印
E (50041) uart: uart_write_bytes(775): uart driver error

diylxy, 2025-07-12:
> E (6567) uart: uart_driver_install(927): uart rx buffer length error(>128)
> 
> 打开TCP发送任意数据时串口打印 E (50041) uart: uart_write_bytes(775): uart driver error

@RaynorJoel [这种方法](https://github.com/windowsair/wireless-esp8266-dap/issues/92#issuecomment-3064865678)应该可以解决你说的问题。在ESP8266上测试通过。

commit bb6ccedf (2025-07-14): fix(esp8266): correct UART port assignment for serial initialization

ESP8266 only supports UART1 as TX-only. Previously, the code attempted to use UART1 as RX and UART0 as TX, causing initialization failure. This commit swaps the UART ports, assigning UART1 as TX and UART0 as RX, resolving the issue. fix #83 and #92
closed 2025-07-14 by windowsair