windowsair/wireless-esp8266-dap #87: DAP status led feature and UART pins

MihailDenisov, 2025-03-26:
Good day!
Many thanks to the authors for developing this product!
1. Is it planned to implement the LED indication function of the DAP status in the future? The LED_CONNECTED and LED_RUNNING output function would be useful for visual control of the device operation.
2. Is it possible to move the UART pins from contacts 18 and 19 to contacts 20 and 21 in the ESP32C3? The fact is that on some modules (supermini) these contacts of the microcircuit do not have contact pads on the board and are connected directly to the TYPE-C connector.

windowsair, 2025-03-27:
Hi, @MihailDenisov 

> Is it planned to implement the LED indication function of the DAP status in the future? The LED_CONNECTED and LED_RUNNING output function would be useful for visual control of the device operation.

This feature was implemented early on and is now intentionally removed because I feel there are few scenarios for its use.

> Is it possible to move the UART pins from contacts 18 and 19 to contacts 20 and 21 in the ESP32C3? The fact is that on some modules (supermini) these contacts of the microcircuit do not have contact pads on the board and are connected directly to the TYPE-C connector.

Yes, you can do this with IOMUX on all espressif chips.

MihailDenisov, 2025-03-28:
> Hi, [@MihailDenisov](https://github.com/MihailDenisov)
>
>
> This feature was implemented early on and is now intentionally removed because I feel there are few scenarios for its use.
>
>
> Yes, you can do [...]

Ok, I'll experiment with this.