windowsair/wireless-esp8266-dap #88: gpio_struct.h: No such file or directory

buzmeg, 2025-04-10:
When I try to compile with "idf.py build", I get an error.

This occurs for both tagged v0.3.0 as well as master.

```
/var/home/foo/esp32dev/wireless-esp8266-dap/components/DAP/include/gpio_common.h:43:14: fatal error: soc/esp32s3/include/soc/gpio_struct.h: No such file or directory
   43 |     #include "soc/esp32s3/include/soc/gpio_struct.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1011/1061] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj
ninja: build stopped: subcommand failed.
```

```
% idf.py --version
ESP-IDF v5.4.1
```

windowsair, 2025-04-10:
No support v5 esp-idf. Use v4.4.2

buzmeg, 2025-04-10:
v4.4.2 is 3 years old at this point.

Is there a timeline on an upgrade?